Limine-Bootloader/Limine #257: Loading .iso or (.tar, .rar?/.zip?)

N1NJ4, 2023-03-13:
Hello, I've ready that you support ISO, and somewhere I saw tar was supported, so I've been trying to load a few though no luck.

basically I've tried these;

```
PROTOCOL=chainload
PROTOCOL=multiboot2
```

with;
```
IMAGE_PATH=fslable://LABEL/file.iso
```

though no luck, either asks for KERNEL_PATH or, from memory trying various things that probably weren't correct... says the PATH isn't correct/findable

haven't found anything else that can help, and I'm not really interested in unpacking them, I'm actually having some trouble loading some unpacked ones on their own partitions. They work from windows boot manager as they are, though loading them.. I haven't found anything to work of yet.

braindigitalis, 2023-03-13:
did you use fslabel or fslable? because your example you give contains a typo and if you used that in the config file it isn't going to work properly...

czapek1337, 2023-03-13:
Chainloading am ISO or other type of archive won't work anyway. You can only chainload other boot sectors on BIOS or other EFI executables on UEFI.

SniperYagi, 2023-03-14:
> Hello, I've ready that you support ISO, and somewhere I saw tar was supported, so I've been trying to load a few though no luck.
>
> basically I've tried these;
>
> ```
> PROTOCOL=chainload [...]

An example to load an ISO would I believe be like this:

DEFAULT_ENTRY=1
TIMEOUT=5

:ISO Loader
     PROTOCOL=limine
     IMAGE_PATH=boot:///<path/to/iso>
     
     
This is just an example though

mintsuki, 2023-03-14:
Limine *cannot* boot ISO files. This isn't as simple as people coming from Ventoy have come to think it is. You need to memory map and emulate a (optical or otherwise) drive.

Ventoy contains a lot of special handling and spaghetti code to allow this to work on/with every OS and bootloader, as that's what Ventoy does. Limine is not Ventoy and their scopes are completely different.

closed 2023-03-14 by mintsuki