Limine-Bootloader/Limine #266: Limine Crashes on startup by just holding any key

Nokodev-programming, 2023-04-22:
To trigger the crash. Start the os and right before the limine bootloader ends and the os kick's in. Hold any key. This will result to a crash (CPU will go shutdown state)

Original Issue located at https://github.com/CosmosOS/Cosmos/issues/2651

mintsuki, 2023-04-22:
Hello, thanks for the report.

Unfortunately, I cannot reproduce this issue. Can you be more specific regarding the steps to reproduce the problem?

Nokodev-programming, 2023-04-22:
When you boot a os with limine. When seeing the limine bootloader's screen, hold any key and it will crash

Video of demostration (Don't ask about the wallpaper background as thats a joke wallpaper)
https://streamable.com/0v6r3h

mintsuki, 2023-04-22:
I still cannot reproduce. Make sure that you are using the latest Limine version and try again and let me know.

Nokodev-programming, 2023-04-22:
Im using Limine from COSMOS. I currently don't know what version of Limine they are using. (Also i am using someone's elso operating system) I am not sure if limine has fixed this or not. I will try to send another video with the keyboard inputs im sending. Maybe will that help out.

mintsuki, 2023-04-22:
I doubt it is going to be helpful. Tell upstream to update the Limine version to latest.

Nokodev-programming, 2023-04-22:
It seems like VMPlayer will prevent any onscreen keyboard software from detecting keys. What i do know is cosmos started using limine this year. The seccond month of 2023. So maybe the latest version of limine of that date.

mintsuki, 2023-04-22:
Yes, which is definitely not the latest version. I know because I could see an issue from the video you sent that has been fixed semi-recently. Again, please tell Cosmos or the OS vendor/developer(s) to update Limine versions.

Nokodev-programming, 2023-04-22:
It seems like that the entire server went inactive. I am not sure if they gonna response as in central europe time its less than a hour away from being night.
[image]

mintsuki, 2023-04-22:
Running `strings <Image file path here> | grep 'Limine 4\.2' | uniq` should show the version of Limine.

Nokodev-programming, 2023-04-22:
Limine 4.20221216.0

mintsuki, 2023-04-22:
The latest version is 4.20230422.0 (actually released today).

Nokodev-programming, 2023-04-22:
Let me check some diffirent iso's from the community to comfirm its Limine 4.20221216.0 and if its somewhere in the range. il close this issue and attempt to ask the cosmos devs to update to the latest version of limine

Nokodev-programming, 2023-04-22:
It looks like the limine's version from cosmos is 4.20221216.0. Il go ask them and close this issue

closed 2023-04-22 by Nokodev-programming