Limine-Bootloader/Limine #279: Scroll on the builtin console

mustevenplay, 2023-06-23:
I was just debugging the boot of a new installation, unsure of how to access the EFI image from a disk other than the one in which Limine is installed onto. My attempts to make it work have been unsuccessful until now.
In the process, I tried to see what the Limine's builtin console could help me with, or at least what I could do with it.
There's an interesting command in it, that I think could aid me, "lsvol". Running it I can see there's information with which I could do something to fix my problem, but I found another: **I can't scroll up in the console's screen.** I tried using the arrows, vi-like key mappings, the PgUp and Home keys, and other types of key combinations.

I'm not really sure how I could make that command (or rather, the console itself) actually useful in this case, since I basically have to see and guess what could be what.

Maybe it's a keymap issue, so my non-english keyboard can't "reach" the specific key to do that? Is it maybe that it doesn't have that functionality implemented yet?

mintsuki, 2023-07-09:
Hi. In commit 11f50a6aaa6b5998be71406954265ed42d48ad88 I've made it so you have to press a key between each entry displayed by `lsvol`.

I hope this satisfies this request :)

closed 2023-07-09 by mintsuki

mustevenplay, 2023-07-09:
Thanks! I just managed to fix the issue I was having! :)