Limine-Bootloader/Limine #283: Add a command to source/include another config file

mekb-turtle, 2023-07-23:
I think it would be nice to be able to include other config files in limine.cfg, like how #include works in C

Example:

limine.cfg:
```
%colors.cfg
TIMEOUT=5
... etc ...
```

colors.cfg:
```
TERM_PALETTE=...
TERM_BACKGROUND...
... etc ...
```

Andy-Python-Programmer, 2023-07-23:
```cfg
:Aero
    CMDLINE=$INCLUDE(kernel.toml)
```

Also having something like the above would be useful.

blanham, 2023-07-23:
Also also, it would be nice to include variable names in the include statements. This would be useful, for example, to include different configs for different architectures

mekb-turtle, 2023-09-08:
> ```haproxy-config
> :Aero
>     CMDLINE=$INCLUDE(kernel.toml)
> ```
> 
> Also having something like the above would be useful.

wouldn't that just be the `${variable}=value` thing?

paoloschi, 2024-02-10:
I logged in precisely with the intention of making the same request, and I see with pleasure that I am not alone in finding that the possibility of importing a separately generated file into `limine.cfg` should be very useful.

My use case: 
I would like to make kernel install|remove hooks into '/etc/kernel.d' so that, when package manager acts on `linuxN.N` packages, my script it would keep the whole ___entries___ section of `limine.cfg` always updated whith as many entries for as many OSs and modules I have at my disposal at that moment.
It would be like having the equivalent of `/etc/default/grub`+`grub-mkconfig` but without all the GRUB's bloat...

sagehane, 2024-05-26:
I'd also want a feature similar to this. It would even be nicer if it's possible to add multiple files with one command using some pattern matching.

My use-case is to add support for Limine on GNU Guix in a manner similar to systemd-boot's usage on NixOS.

On NixOS, systemd-boot is currently used like this:
```sh-session
$ cat /boot/loader/loader.conf 
console-mode keep
default nixos-*
editor true
timeout 5

$ tree /boot/loader/
/boot/loader/
├── entries
│   ├── nixos-generation-248.conf
│   ├── nixos-generation-249.conf
│   ├── nixos-generation-250.conf
│   └── nixos-generation-251.conf
├── entries.srel
├── loader.conf
└── random-seed

2 directories, 7 files
```

closed 2025-08-31 by qookei