Limine-Bootloader/Limine #285: Suggestions for the menu

dankcuddlybear, 2023-07-28:
1) When TIMEOUT=0, the default entry is booted immediately, and I am unable to access the menu. Make it so that the default entry still boots immediately, but the menu appears if a key is held when Limine is loaded, allowing you to choose a boot option.
2) Remove  loading messages when loading initramfs/kernel: "Loading boot:///initramfs-linux.img" "Loading boot:///vmlinuz-linux.img" as it looks nicer without these messages
3) Remember previously booted entry

mekb-turtle, 2023-07-31:
For 2, have this when the QUIET=yes option is set

ginzberg, 2024-03-01:
+1 on "remember previous booted entry"
This is the one feature from grub that I miss. For future historians, in grub you can define `GRUB_DEFAULT=saved`, and the behavior will essentially set the default entry upon booting.  As a proposal for how one could configure this in liming, I believe setting `DEFAULT_ENTRY=-1` could potentially serve the purpose.

rlees85, 2024-03-29:
Just started using this bootloader on an old laptop and couldn't be bothered with GRUB. Really like it. Point 1 about being able to access the menu when a certain key (suggest ESC or F8) is pressed would be very useful. Its nice to have no timeout to boot the default option but have a way of booting alternative options to debug for example if the main boot option won't work (kernel update).

chrisdebian, 2025-03-30:
It's year since the last comment. Anyone know if this still needs to be open?

mintsuki, 2025-03-31:
Given that most of the questions were about how to show the menu when booting automatically, this is the idiomatic approach:
```
quiet: yes
timeout: 1
```
This will allow you to press a key within 1 seconds and that will cause quiet mode to be disabled, thus displaying the menu.

closed 2025-03-31 by mintsuki

lzcunt, 2025-03-31:
Menu control could also be exposed to the operating system through the Boot Loader Interface, e.g. allowing to programmatically show or hide the menu on the next boot or control the timeout, which I intend to implement when I have more time.