Limine-Bootloader/Limine #296: Secure boot violation: Limine boots unsigned kernels

alicavus, 2023-08-28:
I have dell Vostro machine with Secure Boot in Deployed mode, only signed executables are alowed. I also enrolled MIicrosoft's certficate, so I have dual boot support and no need to sign official Debian kernels.

I have signed my Limine EFI and I can boot without issue.

Today come in my mind to check untrusted kernels and executables. I just grab kernel and initramfs from monthly release of Arch Linux and put them in my disk. It booted from it. I checked other unsigned kernels - they aso are loading without any issue.

mintsuki, 2023-08-28:
This is by design. If you want to ensure only approved kernels run, as per PHILOSOPHY.md, you need to add the blake2b checksum to the config file (for any file you want, not just the kernel, also initrd or even a wallpaper file), then use the `limine enroll-config` tool to tell the Limine executable which config file to trust, and *then* you can sign Limine.

alicavus, 2023-08-28:
I understand the concept with b2sum verification of files.
It seems we allways have to set ```EDITOR_ENABLED``` to ```no``` if we want Secure boot.

mintsuki, 2023-08-28:
Yeah, pretty much.

mintsuki, 2023-08-28:
I will make it so `EDITOR_ENABLED` will default to `no` if there is a config blake2b enrolled.

commit 2542a28e (2023-08-30): config: Disable editor by default if config b2sum present. Fixes #296
commit 4a46597f (2023-08-30): config: Disable editor by default if config b2sum present. Fixes #296
commit 57e4ae6e (2023-08-30): config: Disable editor by default if config b2sum present. Fixes #296
closed 2023-08-30 by mintsuki