Limine-Bootloader/Limine #302: Allocation failure: trying to chainload Linux on aarm64

dimaqq, 2023-09-11:
I had a working environment with EFI_STUB Linux 6.4.0+, root UUID and initrd.

I've tried to upgrade it to Limine to allow me to switch kernels easier...

However, when I get the syntax right (I think), I end up with:

```
PANIC: chainload: Allocation failure
Stacktrace:
  [0xbc12ceac] <panic+0xb4>
  [0xbc129ebc] <efi_chainload_file+0x230>
  [0xbc129c8c] <chainload+0x6c>
  [0xbc126474] <boot+0x1d8>
  [0xbc125ec8] <_menu+0xd64>
End of trace. Press a key to return to menu.
```

Environment:

UTM on Mac

```
>>> version
5.20230909.0
```

Manually edited entry that triggers this error:

```
?PROTOCOL=chainload                                                                ?
?IMAGE_PATH=boot:///Image                                                     ?
?CMDLINE=root=UUID=UUID=4c3e28ab-713f-4718-8a9b-434440560ce9 rw                ?
?MODULE_PATH=boot:///initramfs-linux.img  
```

The files exist, `Image` is only `42889728` in size.

mintsuki, 2023-09-11:
Did this also happen on `5.20230830.0`?

dimaqq, 2023-09-11:
Sorry I'm not sure I have a good way to test.
I can try tomorrow maybe.

qookei, 2023-09-11:
How much RAM did you assign to the VM?

qookei, 2023-09-11:
Also worth noting, `MODULE_PATH` doesn't work with EFI chainloading, you need to specify the path to the initramfs on the kernel command line.

mintsuki, 2023-09-11:
The issue should be fixed as of 277cec8, please reopen if that's not the case.

closed 2023-09-11 by mintsuki

dimaqq, 2023-09-12:
Thank you! :)
I'll retest a bit later when arch linux catches up or maybe I'll build from source.

commit 277cec6c (2023-09-11): protos/chainload: Fix bug that could result in bogus oom errors. Fixes #302
commit 7d527d86 (2023-09-11): protos/chainload: Fix bug that could result in bogus oom errors. Fixes #302
commit b54c3fc9 (2023-09-11): protos/chainload: Fix bug that could result in bogus oom errors. Fixes #302