Limine-Bootloader/Limine #304: Hybrid Setup can not boot old kernel on old pc

imrn, 2023-09-16:
1) Working setup (Non-uefi old  pc):

USB flash drive + gpt + Syslinux +
Linux kernel WITHOUT stub and handover capability

2) Partially working, non-working setup:

USB flash drive + gpt + 
lilime with both bios/gpt (created with "imine bios-install") and x86 uefi capability. (limine-bios.sys is on the disk)

This hybrid setup boots:
-  A kernel WITH stub and handover capability, both on an old pc and on an uefi pc. Perfect!

- But it can not boot a kernel (the one at setup (1)) WITHOUT stub and handover capabillty on an old pc.

I would expect booting this old kernel on an old non-uefi old pc, just like setup (1).

Thank you.

imrn, 2023-09-16:
Some more info:

I've compiled the kernel with all necessary options for an efi boot. However, it still does not boot.

But it seem initrd is the trick.

Above non-uefi kernel is a self contained one capable of booting without any initrd. At least that is case under syslinux.

However, under limine that kernel always needs an initrd to boot. Providing a minimal initrd helps to boot on a non-uefi pc.

imrn, 2023-09-17:
Still more info:

Above initrd requires video driver (i915 in my case) to boot under limime. Otherwise, system goes dark and freezes after loading kernel and initrd. (System also goes dark when booting without initrd)

It seems that freeze is not just an screen issue. Ctrl+alt+del or any other thing does not help for anything.

Under syslinux a non-fancy kernel without uefi stuff, without initrd and and without video driver, etc just boots smoothly in vga mode.

(Now, I'm also editing above comments to be precise.)

mintsuki, 2023-09-17:
Hello, sorry for the semi-late reply.

Have you tried passing `TEXTMODE=yes` in the Linux boot menu entry and seeing if that helps?

In case that helps, I assume the kernel gets compiled without linear framebuffer support under the assumption that every BIOS loader will drop Linux into text mode, and this is not the case with Limine.

imrn, 2023-09-17:
Thank you for the reply.. ;)

"textmode=yes" did not help booting without video driver. It just freezes as before. Hence, booting is not possible boot without initrd.
(Kernel has framebuffer and uefi support now.)

With initrd and limime, this kernel can boot on uefi and non-uefi pcs.

But, without initrd (thus without video driver) it can only boot under syslinux.

imrn, 2023-09-17:
I've built this generic hybrid flash drive with this compact and static kernel as a generic tool for booting any uefi and non-uefi pc.

It would be a big plus if limine can be independent of any video driver and initrd.

mintsuki, 2023-09-17:
if you could provide me with the kernel file for me to run a couple of tests with, I'd appreciate it.

There is nothing in principle preventing Limine from booting kernels without video drivers, especially if syslinux boots them fine. Unless this is some sort of regression in Limine, Linux, both, or something wrong with your kernel and/or setup.

mintsuki, 2023-09-17:
If you have trouble sending the kernel file over due to GitHub, try making a gzipped file out of it, that should do the trick.

imrn, 2023-09-18:
I guess this is a minimal kernel that can boot a common intel pc or laptop without needing any module.  It can mostly be used without any initrd.  Other non-critical moldules can be loaded after booting.

I'm including kernel, modules and config.

[config-6.5.3_92-dirty.zip](https://github.com/limine-bootloader/limine/files/12645222/config-6.5.3_92-dirty.zip)
[vmlinuz-6.5.3_92-dirty.zip](https://github.com/limine-bootloader/limine/files/12645224/vmlinuz-6.5.3_92-dirty.zip)
[modules-6.5.3_92-dirty.zip](https://github.com/limine-bootloader/limine/files/12645229/modules-6.5.3_92-dirty.zip)

imrn, 2023-09-18:
One more thing;

After booting with limine, setfont seems to load console fonts. But keys with special characters (ığüşiöçIĞÜŞİÖÇ) display erratic output on console. Loading fonts again does not correct the issue either.

I'm not sure but I guess limine modifies some console characteristics so that font loading mechanisms are garbled.

Booting same kernel and initrd with syslinux again does not have such problem.

Although limine provides nicer look, messng with console and video is not a good idea.

After all, a bootloader only needs to load kernel, related payload, and provide boot options. Fancy things turn into problems.

Currently, I'm not very sure about the nature of the problem, but I,ll try to provide more detail.

mintsuki, 2023-09-18:
Alright, so, I have done a couple of tests using the kernel you provided and I cannot reproduce any of the claimed symptoms.

Limine does not alter the video ROM text mode characters in any way, so about the issue you're describing, I have no idea. And also Limine *has* to "mess with console and video" in order to change video modes, but that has never caused any issue similar to what you're describing to me or anybody else to my knowledge.

What system is this being tried on? Is there any way you could put together and send a disk image with everything required to provide a minimal reproducible example? That would be very helpful.

imrn, 2023-09-19:
Ok. I'll simplify, pack and send the usb boot image. But that may take sometime. But, I think following procedure is more efficient and precise.

So you've:

- Copied modules to /lib/modules
- Copied vmlinuz to efi partition
- Booted it successfully with limine without any initrd and with only kernel option 'root=...'

- Repeat all it with syslinux.

- Repeat all with initrd.

- And repeat all above with a non-uefi pc.

Can you confrm, you do NOT get black frozen screen for any limine cases?

- I consistently get normal booting with syslinux for all of them.

But for limine:
- I can only boot succesfully with an initrd.
- And get black frozen sceen without initrd.

After booting:
- Non-ascii console fonts are garbled for all liimine cases.
- Luckly, ascii chars are fine. So console is usable. 

Configurations:
Uefi laptop: Asus B1402CBA. Intel 1235U.
Non-uefi laptop: Asus 1015 Netbook. Intel N570 Atom.

No fancy, just common Intel machines with only internal Intel GPUs.

imrn, 2023-09-19:
Those all tell me that there is some kind of an iteraction between limine and video driver.

For some cases it works, for some combinations it does not.
Especially, when video driver is absent machines can freeze.

I can understand than combinations are hard to debug. But, such a close ınteraction between a bootloader and video driver does not feel philosophically right.

I hope you would reconsider it.

alicavus, 2023-09-19:
Sometimes buggy BIOS firmware causes strange issues, this is why I suspect BIOS firmware issue.

Did you have latest BIOS firmware?

imrn, 2023-09-19:
Yes. BIOSes are upto-date.
Note that, situation is same and old and new, uefi and non-uefi pcs.

I think given the kernel and above definition is prety stable and consistent way of describing the problem. Even better than boot image because you can try the combinations more easily in your controlled environment.

Just copy the kernel and modules to their places and add an boot entry to limine conf file.

Did you try it?

Then generate an initrd and give it a try again.

I'll be happy to hear you results..

mintsuki, 2023-09-19:
> Ok. I'll simplify, pack and send the usb boot image. But that may take sometime. But, I think following procedure is more efficient and precise.
> 
> So you've:
> 
> * Copied modules to /lib/modules
> * Copied vmlinuz to efi partition
> * Booted it successfully with limine without any initrd and with only kernel option 'root=...'
> * Repeat all with syslinux.
> * Repeat all with initrd.
> * And repeat all above with a non-uefi pc.
> 
> Can you confrm that you do NOT get black frozen screen for limine cases?
> 
> * I consistently get normal booting with syslinux for all of them.
> 
> But for limine:
> 
> * I can only boot succesfully with an initrd.
> * And get black frozen sceen without initrd.
> 
> And after booting:
> 
> * Non-ascii console fonts are garbled for all limine cases.
> * Luckly, ascii chars are fine. So console is usable.
> 
> Configurations: Uefi laptop: Asus B1402CBA. Intel 1235U. Non-uefi laptop: Asus 1015 Netbook. Intel N570 Atom.
> 
> No fancy, just common Intel machines with only internal Intel GPUs.

I can confirm that the kernel gets booted until the point where it panics because no root is provided. This tells me there probably was no issue on Limine's end booting the kernel you provided. This is why I need a reproducible image containing things down to the rootfs and init that you used.

> Those all tell me that there is some kind of an iteraction between limine and video driver.
> 
> For some cases it works, for some combinations it does not. Especially, if video driver is absent during the boot time machines can freeze.
> 
> I can understand than combinations are hard to debug. But, such a close interaction between a bootloader and video driver does not feel philosophically right.
> 
> I hope you would reconsider it.

Limine does not interact with "the video driver" as that's not even loaded or loadable before the OS does so long after Limine has passed control over. Nothing to reconsider here.

> Yes. BIOSes are upto-date. Note that, situation is same on old and new, uefi and non-uefi pcs.
> 
> I think given the kernel and above definition is prety stable and consistent way of describing the problem. Even better than boot image because you can try the combinations more easily in your controlled environment.
> 
> Just copy the kernel and modules to their places and add a boot entry to limine conf file.
> 
> Did you try it?
> 
> Then generate an initrd and give it a try again.
> 
> I'll be happy to hear you results..

No I did not try it because I was sleeping, and now that I am awake I do not feel like wasting a bunch of time trying to recreate an image that would display this problem when you, the person who opened this issue, would know best and easily be able to put it together as I asked.

So please send an image file displaying the issue in a reproducible way (on your end).

imrn, 2023-09-19:
> I can confirm that the kernel gets booted until the point
> where it panics because no root is provided.

Thank you for providing some result after you've requested the kernel.

On my side, I have complete blackout after 'loading kernel..' message , following the limine boot menu. Interestingly only providing an initrd makes it possible to boot. Graphics=no, Verbose=yes, textmode=yes, etc has no effect.

> No I did not try it because I was sleeping, and now that I am
> awake I do not feel like wasting a bunch of time trying to recreate
> an image that would display this problem when you, the person
> who opened this issue, would know best and easily be able to
> put it together as I asked.

Only below command is sufficient for generating an initrd:
(For dracut I guess it something similar)

mkinitcpio -k kernel -g initrd.img

> So please send an image file displaying the issue in a
> reproducible way (on your end).

Thanks for your help. Mature software needs patience. I was willing to use limine and provide feedback as I saw it promising. But as I understand it needs more work and time to get mature. 
And attitude matters.

Anyway, for now I decided to go back to good old syslinux. Good luck..

closed 2023-09-19 by imrn

mintsuki, 2023-09-19:
Sorry that you refused to cooperate for some reason, and that you somehow think me running `mkinitcpio` on a completely different host OS is going to generate the same initrd as yours, or that somehow I can pull a rootfs out of a hat, all because you refuse to send over an image with all of this set up in the way that causes the misbehaviour for you.

It seems to me like you were here to complain more so than to get the issue fixed, and that is not welcome.