Limine-Bootloader/Limine #305: Secure boot "Verification Failed" error since v5.20230909.0 on Arch

sonnysighedup, 2023-09-18:
Since release v5.20230909.0, I've been unable to use secure boot successfully with limine. I'm on an Arch-based variant, and up through v5.20230830.0 limine has worked flawlessly for me with secure boot enabled.

My updating process after each limine pkg upgrade:
1. Manually copy the updated BOOTX64.EFI to the appropriate place on my efi partition, calling it limine.efi
2. Signing limine.efi with sbsign
3. Enrolling the blake2b sum of my limine.cfg into limine.efi
4. Re-signing limine.efi with sbsign

What I have noticed in these latest upgrades and subsequent secure boot errors is that I no longer see a warning message like
`warning: data remaining[241592 vs 241664]: gaps between PE/COFF sections?`
after each instance of signing limine.efi

Any ideas on what might have changed from v5.20230830.0 that I need to adjust to?

Thank you for this great bootloader!

mintsuki, 2023-09-18:
Hi, thanks to you for the valuable feedback!

Is this by using `shim` or is this by directly signing Limine with a key you manually enrolled? Does this happen with `5.20230917.0` (I assume it still does)?

If this is by using `shim`, I suggest trying without it since Limine doesn't officially provide support for it and may be the source of issues like these.

sonnysighedup, 2023-09-18:
Yes, I'm directly signing Limine with a manually enrolled key. I use `sbsign` from `sbsigntools`. My typical command is:
`doas sbsign --key /etc/efikeys/mok/MOK.key --cert /etc/efikeys/mok/MOK.crt --output /efi/EFI/BOOT/limine.efi /efi/EFI/BOOT/limine.efi`

I did try with `5.20230917.0` and still had the issue. Once I turn off secure boot, the booloader works. Downgrading to `5.20230830.0` and then re-enabling secure boot from BIOS makes everything goes back to normal.

mintsuki, 2023-09-18:
Do you have any idea whether the "Verification Failed" error comes from the Linux kernel, firmware/BIOS, or else?

mintsuki, 2023-09-18:
Because as far as I can tell "Verification Failed" seems to be a `shim` error, so please make sure you're not using shim.

alicavus, 2023-09-18:
@sonnysighedup  why you are signing your bootloader twice?

sonnysighedup, 2023-09-18:
> @sonnysighedup why you are signing your bootloader twice?

An unnecessary precaution. I don't where I got the idea or developed the habit from. As it works either way, I will wean myself from the unnecessary extra step. Thanks!

closed 2023-09-18 by sonnysighedup

mintsuki, 2023-09-18:
No worries! I am just glad you sorted your issue out in the end.