Limine-Bootloader/Limine #307: hibernation fails to resume

Pound-Hash, 2023-10-05:
With grub, I could successfully hibernate via:
```
echo shutdown >| /sys/power/disk
echo disk >| /sys/power/state
```
But this doesn't work with limine. The system shuts down, but fails to resume. I'm using the linux protocol.

Limine 5.20230811.0
Gentoo

mintsuki, 2023-10-05:
Hello, thanks for the report.

Do you have any relevant log associated with the issue? Anything from `dmesg` that seems related to the problem?

Could you try the latest version (5.20230928.2) and see if that exhibits the problem as well?

Pound-Hash, 2023-10-09:
Thanks for the prompt reply. I figured it out. The problem was with dracut:

https://bugzilla.redhat.com/show_bug.cgi?id=2060501
https://news.ycombinator.com/item?id=28644762

closed 2023-10-09 by Pound-Hash