Limine-Bootloader/Limine #308: One-Time Boot

CocoXiaoLong, 2023-10-08:
I have Linux and Windows installed and I want to boot into Linux all the time and into Windows only when I need it.
Is it possible to make config file which the bootloader take a single time to boot and deletes it afterwards?
So I can copy this file to the bootdrive in Linux and reboot into Windows and then only need to reboot to get back to Linux?
The problem with just editing the main config, is the `I can't go back` factor. Once in Windows, it would be pain to mount the boot partition and edit it back.
I'm not familiar with the codebase or if it's even possible to edit files permanently.

qookei, 2023-10-08:
This functionality is already implemented by UEFI, via the `BootNext` variable. You can manipulate the variable both from Linux (via `efibootmgr`) and from Windows (via `bcdedit`):

Linux:
To see a list of all boot entries: `efibootmgr`
To set BootNext to a specific entry (replacing `XXXX` with it's number): `sudo efibootmgr --bootnext XXXX`

Windows:
To see a list of all boot entries: `bcdedit /enum firmware`
To set BootNext to a specific entry (replacing `UUID` with the one from the previous command): `bcdedit /set {fwbootmgr} bootsequence {UUID}`

CocoXiaoLong, 2023-10-08:
That's the problem, you might've guessed it: My PC uses MBR...

mintsuki, 2023-10-08:
Assuming by "MBR" you mean legacy BIOS, then no, it cannot be done, sorry.

closed 2023-10-08 by mintsuki