Limine-Bootloader/Limine #309: KERNEL_PATH - The URI path of the kernel.

fungilife, 2023-10-09:
an .elf file path a URI path ... 

Back in March 23 when I started packaging and offering limine I tried it and it worked (after a long hastle with guess work of making a working limine.cfg)  Now I have a user asking what is wrong and the nightmare of guesswork returns.

Anything I try on this 

    KERNEL_PATH=

is rejected

Why can't some examples of a general linux use can be offered to remove the guesswork of what MAY some of this tiny documentation mean by "terms".
Your site leads to download, no documentation, the git, and some social-media server that is can not and will not be accessed.  Is this where meaningful documentation is stored?  In a chatroom server?

I have done more than 20 upgrades of limine ... I believe I should be able to provide users with some guidance in making it work

ElectrodeYT, 2023-10-09:
Consider reading [this](https://github.com/limine-bootloader/limine/blob/trunk/CONFIG.md)

Basically, if you want to read a elf file from the boot partition, its `boot:///path/to/elf/file`

no92, 2023-10-09:
[There is documentation on this](https://github.com/limine-bootloader/limine/blob/trunk/CONFIG.md#uris). This documentation is referenced in the README and easily found in a clone, too. I don't know what the ranting is saying or trying to achieve.

mintsuki, 2023-10-09:
All the necessary documentation to write a working config file is provided in [here](https://github.com/limine-bootloader/limine/blob/trunk/CONFIG.md) as previously mentioned.

There is no such thing as a "general Linux use". It is all very distribution and setup dependent, and giving random magical instructions that will work everywhere is not possible.

Also quit ranting and acting so arrogantly. I can choose which platform I want to use to communicate with the community, and if you don't like it, don't use it, but don't come whining about it in GitHub issues (which by the way also work perfectly well to ask questions without acting like a prick).

That said, if you have an *actual* question, whether a valid bug or just asking for help with a non working configuration, ask said question. Because in this issue's message no question was asked, there was only whining.

lukflug, 2023-10-09:
> Your site leads to download, no documentation, the git, and some social-media server that is can not and will not be accessed. Is this where meaningful documentation is stored? In a chatroom server?

The documentation *does* exist and *is part of* the Git repository.
The Discord server *can* be accessed and *is* accessed.
There's nearly 500 people there!

fungilife, 2023-10-09:
> [There is documentation on this](https://github.com/limine-bootloader/limine/blob/trunk/CONFIG.md#uris). This documentation is referenced in the README and easily found in a clone, too. I don't know what the ranting is saying or trying to achieve.

The term "ranting" is a hostile term for telling another party that you don't make sense, and it would have been better to ask to explain what YOU don't understand.

lukflug, 2023-10-09:
You were kinda being hostile in your original message. And "ranting" doesn't mean that the other party doesn't make sense. It's just that it's an unnecessarily lengthy diatribe.

fungilife, 2023-10-09:
> You were kinda being hostile in your original message. And "ranting" doesn't mean that the other party doesn't make sense. It's just that it's an unnecessarily lengthy diatribe.

If you could put half the energy of being ironic to someone who is asking for help in figuring out your software as to writing maybe "a second page with examples" of how this thing works, maybe there would have been no need for a "diatribe" 

Linux x86_64 dos partitioning mbr partition: say sdc1  on drive sdc
/boot/vmlinuz-linux
/boot/limine/limine.cfg
limine installed already and menu appearing

Write an example bootable entry and if this doesn't work we can proceed to debugging what is wrong.  Till then, the documentation is insufficient

fungilife, 2023-10-09:
> Consider reading [this](https://github.com/limine-bootloader/limine/blob/trunk/CONFIG.md)
> 
> Basically, if you want to read a elf file from the boot partition, its `boot:///path/to/elf/file`

consider that the title of the ISSUE is copy paste from that page, is there another?

fungilife, 2023-10-09:
> You mention ext4... is the `boot/vmlinuz` file on a ext4 partition?
> 
> Limine no longer supports ext4 partitions.

Is there a link to the release that specifically says this, all distro wikis that talk about limine speak of ext4 support when others don't.
And by lack of support does it mean what used to boot with limine and ext4 on MBR will no longer be able to, or just that if it fails don't bother limine devs?  There is a gray area here in the interpretation of the word support!

fungilife, 2023-10-09:
> You mention ext4... is the `boot/vmlinuz` file on a ext4 partition?
> 
> Limine no longer supports ext4 partitions.

https://github.com/limine-bootloader/limine/blob/trunk/README.md#supported-filesystems

Maybe the devs need to make their mind up instead of going back and forth, is it or is it not supported?

closed 2023-10-13 by mintsuki

Pimpdaddy1274, 2023-10-19:
> You mention ext4... is the `boot/vmlinuz` file on a ext4 partition?
> 
> Limine no longer supports ext4 partitions.

I didn't see anything official that shows Limine dropped EXT filesystems support. Unless I missed it at a point and time