Limine-Bootloader/Limine #310: Linux protocol support for aarch64 and riscv64

rsepassi, 2023-10-12:
I see that there's a todo to support the Linux protocol for aarch64 and riscv64.

I believe this is the relevant Linux documentation:
https://docs.kernel.org/arch/arm64/booting.html
https://docs.kernel.org/riscv/boot.html

I wonder if Linux's EFI stub functionality could be leveraged here, where maybe for now the Limine Linux protocol just jumps to the built-in EFI stub (CONFIG_EFI_STUB=y in Linux's default make defconfig):
https://docs.kernel.org/admin-guide/efi-stub.html

rsepassi, 2023-10-13:
> To be honest that doesn't sound like a bad idea!

Sweet! Let me know if I can help. I’m not so familiar with these parts, but
might be a useful hand with some pointers.

mintsuki, 2023-10-13:
Limine already support EFISTUB :)

You just need to use the `chainload` protocol as described in CONFIG.md to load a Linux kernel image as an EFI executable.

rsepassi, 2023-10-13:
Oh sweet! I’ll try that out today and report back.

rsepassi, 2023-10-13:
Worked great on aarch64! Thank you!

```
:simplelinux
  COMMENT=simplelinux
  PROTOCOL=chainload
  CMDLINE=initrd=initrd console=ttyS0
  IMAGE_PATH=boot:///kernel
```

However for riscv64, I can't quite get things working because qemu-system-riscv64 doesn't seem to use UEFI firmware by default and instead uses OpenSBI and things seem to just hang. There also doesn't seem to be readily available UEFI firmware for riscv (there's this but I can't quite tell: https://github.com/riscv-admin/riscv-uefi-edk2-docs).

If you have any ideas for how to test riscv, let me know. I'll close this though since things seem to work great on Limine's side.

closed 2023-10-13 by rsepassi