Limine-Bootloader/Limine #311: no nvme support

greyltc, 2023-10-13:
limine can't see my nvme drives

48cf, 2023-10-13:
Can you provide more details? What Limine version are you using? Are you using BIOS or UEFI? How did you come to the conclusion that Limine can't see your drives? Limine itself does not implement support for any storage, it's generally provided by the firmware, so as long as your platform is able to boot from the drive without Limine, the issue is either misconfiguration or firmware reporting invalid/unexpected information that Limine can't work with. Please ensure the drive is not recognized by entering the console (press the C key in the Limine boot menu), typing "lsvol" and verifying that it is not listed there. You can do that by booting into an operating system and checking the UUID of a partition on the affected drive, then verifying it isn't in the console output.

mintsuki, 2023-10-13:
Yes, correct. Limine inherits storage device drivers from the firmware, so in case NVMe drives are not detected, that sounds like some firmware issue or potential firmware/Limine misconfiguration.

Closing this issue as invalid. Additionally I'd like to point out that I use Limine myself on many NVMe devices and it works just fine :)

greyltc, 2023-10-13:
I'm on the latest version v5.20231006.0 as of this writing. I'm not sure what additional details to give, `lsvol` in the limine shell pretends like the nvme drives don't exist.

If I switch to another bootloader, [clover](https://github.com/CloverHackyColor/CloverBootloader), that one can see my nvme drives, so I'm not sure how this issue can be invalid since other bootloaders have solved it.

mintsuki, 2023-10-13:
This issue is invalid because Limine does not specifically care about the fact that certain storage devices may be NVMe or not, and because the issue was originally titled "no nvme support" which is indeed invalid, and no real substantial details to go off of were provided.

The output of `lsvol` and the specifics about your computer would be helpful information to give, to start off.

mintsuki, 2023-10-13:
I reopened the issue and changed it from invalid to unconfirmed following your edit to the title and change in scope of the issue.

greyltc, 2023-10-13:
Apologies for the bad title scope. In my own little world, 100% of the NVMe disks I asked limine to detect were not found, so I guessed that it lacked support in general. Requested details forthcoming...

greyltc, 2023-10-13:
The machine is a Dell Inc. PowerEdge T630.

The drives are
```
├─Timetec 35TTFP6PCIE-256G:
│     Device ID:          [REDACTED]
│     Summary:            NVM Express solid state drive
│     Current version:    V0530B0
│     Vendor:             Silicon Motion, Inc. (NVME:0x126F)
│     Serial Number:      [REDACTED]
│     GUIDs:              [REDACTED] ← NVME\VEN_126F&DEV_2263
│                         [REDACTED] ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
│                         [REDACTED] ← Timetec 35TTFP6PCIE-256G
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
├─MZVLB256HAHQ-000L7:
│     Device ID:          [REDACTED]
│     Summary:            NVM Express solid state drive
│     Current version:    1L2QEXD7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      [REDACTED]
│     GUIDs:              [REDACTED] ← NVME\VEN_144D&DEV_A808
│                         [REDACTED] ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         [REDACTED] ← SAMSUNG MZVLB256HAHQ-000L7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
```
They're visible from Linux once the system boots and from the CLOVER bootlader and otherwise work as expected.
The output of `lsvol` seems to be not so easy to capture as text, so I'll put this image here: 
[image]
The only device/partition that's found there is my boot media (where limine lives).

dermotbradley, 2023-10-25:
Could this be due to Intel Volume Management Device (VMD) being present and enabled on your Dell?

I have seen the issue where machines that have NVME devices attached via VMD either need (a) VMD disabled (via BIOS/UEFI setting), or (b) the "vmd" kernel module loaded by Linux in order to be seem by Linux.

closed 2024-03-09 by mintsuki