Limine-Bootloader/Limine #312: Menu entries now in center of screen

bkauler, 2023-10-14:
EasyOS 5.5.5 has updated to 4.20231006.0 and menu entries are now in the center of the screen. Cannot see any setting for this in CONFIG.md. What if we don't like this change and want the menu back top-left like before?

Note, have posted about this version bump to the Limine thread in the Puppy Forum:

https://forum.puppylinux.com/viewtopic.php?p=100934#p100934

mintsuki, 2023-10-17:
Hello, sorry for the late reply.
No, there is currently no option to restore the previous look; is this something absolutely necessary? Maybe give the new look a shot, I promise it grows on you :p

closed 2023-10-29 by mintsuki

bkauler, 2024-01-18:
It has not grown on me.

I have two different computers that I use regularly, one of them with an older version of Limine with menu at top-left. Comparing, it is quite disconcerting for the eyes to have to move to the center of the screen to see the menu.

There have been many studies done about this. The eyes automatically look at top-left of the screen first, then move left. Here is one link:

https://cxl.com/blog/10-useful-findings-about-how-people-view-websites/