Limine-Bootloader/Limine #316: no changelog, no idea of what's changed from v4 to v5

austanss, 2023-11-05:
One time when I switched from v3 to v4 I ran into compatibility issues with protocol support. After that issue was resolved I learned my lesson to pay attention  to semantic versioning, as I was so laboriously lectured.

Now from v4 to v5 what are the compatibility-breaking issues I need to be aware of? A changelog would do nicely but just an idea of what compatibility breaking changes occurred would satisfy.

austanss, 2023-11-05:
For the record I have not noticed any breaking changes inside my operating system but maybe there is an insidious bug laying dormant somewhere, waiting to confuse me senseless at the worst possible time.

austanss, 2023-11-05:
Whilst I could theoretically compare tags in GitHub, it is rather time-consuming, tedious, and impractical to find two tags that don’t have completely different commit histories (and thus incomparable) and then parse the hundreds of commits to find major breaking changes.

mintsuki, 2023-11-08:
The breaking changes are:

- Some binary files were renamed (for example `limine-cd-efi.bin` -> `limine-uefi-cd.bin` and `limine.sys` -> `limine-bios.sys`).
- The Limine boot protocol terminal feature is no longer supported.
- Support for RISC-V 64-bit is introduced.
- 3 Limine executables coalesced into 1 (`limine-deploy, limine-version, limine-enroll-config` -> `limine`).

closed 2023-11-08 by mintsuki