Limine-Bootloader/Limine #317: Can not boot any UKIs (unified kernel images)

WildPenquin, 2023-11-14:
Hi,

I can not boot my unified kernel image. It seemingly loads the image, but there is an error message (from systemd-boot stub?):

`systemd-boot: Assertion 'prefix' failed at ../systemd-stable/src/boot/efi/stub.c:2680@cmdline_append_and_measure_addons, halting.`

Possibly related: #256. But I'm not using gummiboot, and the error is different.

I'm not setting any cmdline in limine, but (intend to) use the one in the UKI. The same image can boot via systemd-boot and a manual UEFI entry.

Limine configuration:

```
:EFI Chainloading
                COMMENT=Chainload Arch Linux-Zen UKI

                PROTOCOL=chainload
                IMAGE_PATH=boot:///EFI/Linux/arch-linux-zen.efi
                RESOLUTION=3440x1440x32

```

bootctl list entry for the image:
```
         type: Boot Loader Specification Type #2 (.efi)
        title: Arch Linux (6.6.1-zen1-1-zen) (default) (not reported/new)
           id: arch-linux-zen.efi
       source: /boot//EFI/Linux/arch-linux-zen.efi
     sort-key: arch
      version: 6.6.1-zen1-1-zen
        linux: /boot//EFI/Linux/arch-linux-zen.efi
      options: root=UUID=1a5adbc2-4d53-483a-ba58-db298ee1628c resume=UUID=c75a43ee-46ea-4cd3-82e9-da4f73e0da3a rw amdgpu.ppfeaturemask=0xfffd7fff amdgpu.dpm=1 amdgpu.aspm=1 amdgpu.runpm=1 amdgpu.bapm=1 drm.edid_firmware=DisplayPort-2:edid/edid-samcf791-min38.bin initrd=amd-ucode.img initrd=initramfs-linux-zen.img quiet
```
(I haven't tried other UKI images).

mintsuki, 2023-11-14:
Hi, what systemd version is this? I see that the code triggering the assertion you posted in systemd-boot has been heavily modified in recent commits. Could it be possible for you to additionally test whether this also fails with the latest systemd or, if yours is the latest release, could you try building systemd off the main branch and test that and report back?

Thanks.

WildPenquin, 2023-11-14:
Hi,

This is on Arch Linux and systemd 254.6-2.

I will see about compilation, that is certainly something I can do but I'm a bit busy with other work and responsibilities. I'll get back to you when, but probably not in the very next days! =)

I noticed there was a release ~2 hours ago, is that something I should try?

mintsuki, 2023-11-14:
> I noticed there was a release ~2 hours ago, is that something I should try?

Yeah I would do that.

mintsuki, 2023-11-14:
Oh you meant the Limine release? I thought you meant a systemd release. Well I doubt that is going to change much as no chainload related code was touched...

WildPenquin, 2023-11-14:
I quickly installed Arch in a VM and can reproduce the issue there (and succesfully boot the UKIs via other means).

I also installed (or compiled) systemd-git with the help of AUR. The UKIs created with the git version do not boot either (plain default Arch Kernel nor the zen variant). I do get a slightly different error message:

`systemd-boot: Assertion 'prefix' failed at ../systemd-git-stable/src/boot/efi/stub.c:372@load_addons, halting`

The systemctl version:

```
$ systemctl --version
systemd 255 (255.r68797.6060271263-1-arch)
+PAM +AUDIT -SELINUX -APPARMOR -IMA +SMACK +SECCOMP +GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT -QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD +BPF_FRAMEWORK +XKBCOMMON +UTMP -SYSVINIT default-hierarchy=unified
```

mintsuki, 2024-03-09:
This bug was reproduced but caused by systemd-boot, not Limine. Confirmed working with other UKIs. Please report the issue to systemd-boot if it persists.

closed 2024-03-09 by mintsuki