Limine-Bootloader/Limine #320: ext2/3/4 support required

bkauler, 2023-12-16:
We discuss Limine on the Puppy forum:

https://forum.puppylinux.com/viewtopic.php?t=6190

Where member 'luvr' informed us that Limine v6 has dropped support for ext2/3/4.

So, I do hope that v5 is going to be supported indefinitely, as EasyOS requires the bootloader to have ext4 support.

It was sometime ago, that EasyOS implemented a separation of boot and working partitions, requiring that vmlinuz and initrd be in the working partition. This separation policy was introduced in 2022, see blog:

https://bkhome.org/news/202206/complete-separation-between-boot-manager-and-easyos.html

mintsuki, 2023-12-16:
Hi,

What is preventing the kernel and initrd from being put on the "boot" partition, exactly? I can hardly think of situations where that is not something that can be easily done, which is also why this change has been made (see PHILOSOPHY.md).

The kernel and initrd can either be installed there directly or copied there on updates, or this can be achieved in countless other ways...

When it comes to 5.x, if you wish to stick with it, I will probably keep backporting *important* bug fixes for a while, but no more than that. This won't be a 4.x situation where the branch is maintained basically in full in parallel with 5.x.

bkauler, 2023-12-16:
EasyOS used to have vmlinuz and initrd in the boot partition, but a philosophical decision was made to completely separate the installation from the bootloader. As explained in the blog post:

https://bkhome.org/news/202206/complete-separation-between-boot-manager-and-easyos.html

And the one before that:

https://bkhome.org/news/202206/limine-bootloader-for-easyos.html

A typical Puppy Linux user might have a dozen or more different Puppy Linuxes and derivatives such as EasyOS, and other Linux distributions, installed, and it became very messy requiring all the vmlinuz and initrd files to be in the boot partition.

Our decision to completely separate them, makes everything simpler. It is a practical organizational philosophy, especially easy for users to comprehend.

EasyOS used to use rEFInd and Syslinux, but Syslinux requires vmlinuz and initrd to be on the boot partition. So the change in philosophy required that I find something else. That is when I discovered Limine, a much simpler system than grub2.

The bottom line here is that if Limine drops support for ext2/3/4 then I will have to move on, eventually anyway. I will be forced to go for grub2.

bkauler, 2023-12-16:
Regarding 4.x, the last release was tagged as "eol". But it will still receive updates?

mintsuki, 2023-12-16:
Well then, I guess EasyOS/Puppy's and Limine's philosophy are diverging then... ext2/3/4 is unlikely to come back, ever.

If that means that you'll have to move to grub2, so be it. grub2 is definitely a very versatile bootloader so no matter which use case you have in mind, it can probably be made to fit... with all the ups and downs that that versatility and lack of focus bring.

If you want to consider sticking with Limine, may I suggest either reevaluating the necessity to have the kernel/initramfs on an ext2/3/4 partition, or, failing that, there is still the option to keep downstream patches that retain ext2/3/4 support.

closed 2023-12-16 by mintsuki

jamesbond3142, 2023-12-18:
Just a consideration. The default ESP partition in many PCs are limited in size. It is only good for one or two Linuxes; and even then for OS with small initramfs. There are other Linuxes that has huge initramfs that won't fit in the ESP - hence the need to access files located in partitions outside the ESP.

mintsuki, 2023-12-18:
Sure, but I don't see how it is necessary for said additional partition to be ext2/3/4 and not FAT. Also the ESP can be resized. In any case it doesn't seem to me for this to be any more complicated or simpler depending on Limine support for ext2/3/4.

jamesbond3142, 2023-12-18:
Thank you for your reply.

> Sure, but I don't see how it is necessary for said additional partition to be ext2/3/4 and not FAT. 

It is not necessary, I agree, but it is convenient. If the additional partition is formatted ext2/3/4, it can also be used to store Linux stuff in addition to just storing kernels and initramfs-es (instead of yet another data partition). This is especially true for OS with layering support (aka frugal install). 

> Also the ESP can be resized. 

It can, but in many cases it is difficult because usually the ESP and the OS partition is side by side, so resizing the ESP requires one to resize _and_ move the OS partition first (to make space for a bigger ESP), and it gets complicated when in addition to the OS there are other data partitions (which then needs to be resized _and_ moved before the OS partition can be resized and moved, etc). 

> In any case it doesn't seem to me for this to be any more complicated or simpler depending on Limine support for ext2/3/4.

The ability to put Linux data in the same partition with its kernel and initramfs is a convenience. It's not really necessary. But the need to resize and move OS partition to enlarge an existing ESP is a big hassle (and potentially a dangerous one). 

At the end of the day, it's your call. I'm just trying to present another point of view. 

But I have one question, though, if you don't mind. I can understand the argument not to support ext2/3/4 if the support was never there. But Limine already has it, and it's already working. A lot of the hard work has already been done; it's just a matter of maintenance going forward. So why drop it?

mintsuki, 2023-12-18:
I do not have the time nor will to maintain the ext2/3/4 driver, and I do not want Limine's users to have a degraded experience by using a suboptimal driver... I have had all kinds of issue reports concerning ext2/3/4 performance and stability.

Now, given that the ext2/3/4 driver already doesn't align with the Limine philosophy, removing it from a major release was a no-brainer that had always been in the back of my mind.

jamesbond3142, 2023-12-19:
> I have had all kinds of issue reports concerning ext2/3/4 performance and stability. ... I do not have the time nor will to maintain the ext2/3/4 driver, and I do not want Limine's users to have a degraded experience by using a suboptimal driver.

I'm not aware of this. Now that I do, it makes a lot of sense. Thank you for taking the time to clarify the situation.

RaphMar2021, 2023-12-19:
And if you really want ext2/3/4 file system support, you can use Limine 5.X, it's still supported @jamesbond3142