Limine-Bootloader/Limine #322: Add password autentication for local entries

Loara, 2023-12-28:
Linux and Windows operating systems already have an authentication mechanism to prevent unauthorized access to system resources. However a sysadmin might add another protection layer against unauthorized accesses to some entries (for example you want an user to access only some entries and not others like a signed unrestricted UEFI Shell or unstable Linux kernels).

You  could add a `PASSWD_HASH` option in the configuration files that holds a password-hash with salting (for example [yescrypt](https://www.openwall.com/yescrypt/) which is already in libxcrypt) and a `PASSWD_DELAY` option to add a minimum delay after a wrong password.

Another possible feature would be a fail-lock mechanisms that prevents any entry to boot after a configured number of failed attempts held in a configuration record like `PASSWD_FAILLOCK`.

kovmir, 2024-04-11:
> You could add a `PASSWD_HASH` option in the configuration files that holds a password-hash with salting (for example [yescrypt](https://www.openwall.com/yescrypt/) which is already in libxcrypt) and a `PASSWD_DELAY` option to add a minimum delay after a wrong password

That is a really weak security measure. If limine refuses to boot something, I can still do it outside of limine.

Loara, 2024-04-21:
Obviously this feature is meaningful only when SecureBoot is enabled and with BIOS locked.

kovmir, 2024-04-22:
Boot menu is only (is it not?) about having physical access. If I have physical access, I can take the drive out or reset BIOS (UEFI) password.

GitBassador, 2024-04-22:
Still, if your root is unencrypted, dosent that password on the boot menu mean jack shit when someone who is trying to get your data can just open the computer, grab the drive and run?
Thus with standard forensics tools, you can just get whatever you want.
Encrypted root, No acess without your password.
And at that point, theyre gonna come extort the password out of you.
[See relevant xkcd](https://xkcd.com/538)

Loara, 2024-04-22:
It depends, it could be a PC shared among many people and you want only the system administrator to access the UEFI shell for maintenance and avoid unpracticed people to gain access to it.

Clearly this doesn't prevent you to steal the drive and access it, but the main point to have boot entries locked behind a password is to prevent inexperienced people to inadvertently cause troubles to everyone.

Loara, 2024-05-21:
> > to prevent inexperienced people to inadvertently cause troubles to everyone.
> 
> Operating systems provide authentication methods to prevent unwanted access.

EFI shells doesn't have any authentication method. Moreover, Linux kernel can be booted with different kernel parameters and you can't prevent an user to boot only one of specified kernel but not the others without creating a new partition with different users.

GitBassador, 2024-05-21:
Just set EDITOR_ENABLED or other flags you want off to no in the configuration. Relevant config flags list:
https://github.com/limine-bootloader/limine/blob/v7.x/CONFIG.md

kovmir, 2024-05-22:
> EFI shells doesn't have any authentication method. Moreover, Linux kernel can be booted with different kernel parameters

You were talking about `inexperienced people to inadvertently cause troubles to everyone`.

mintsuki, 2024-08-04:
Yeah, I don't think this would ever make it upstream, sorry. Closing.

closed 2024-08-04 by mintsuki