Limine-Bootloader/Limine #325: reserved identifier violation

elfring, 2024-01-20:
I would like to point out that identifiers like “[`_LIMINE_H`](https://github.com/limine-bootloader/limine/blob/c15535d8954c72a08a6c00a8bccf984995d6a3a9/limine.h#L17-L18 "Update candidate")” and “[`__MENU_H__`](https://github.com/limine-bootloader/limine/blob/c15535d8954c72a08a6c00a8bccf984995d6a3a9/common/menu.h#L1-L2 "Another update candidate")” [do eventually not fit](https://wiki.sei.cmu.edu/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier#DCL37C.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28IncludeGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

mintsuki, 2024-01-20:
Thanks for reporting the issue. This is indeed a problem, but one that plagues C codebases all throughout, not just the Limine project...

I have changed the include guard macro in `limine.h` as that is the most important one since it's the exposed header, and changed Flanterm to also not use reserved identifiers for its include macros. I also used `-isystem` instead of `-I` for the `freestanding-headers` includes (as it should be since they are libc-replacement headers).

Other than that, it hardly matters since the reserved identifiers used are unlikely to cause conflicts, but if there are any issues at any point stemming from reserved identifier conflicts I'll fix those on a case-by-case basis.

closed 2024-01-20 by mintsuki

elfring, 2024-01-21:
Thanks for [your adjustment of an affected include guard](https://github.com/limine-bootloader/limine/commit/2afb8e64eb9436682bae11ad21011ed33cd7a72b#diff-6bc642fe6b8d10ed18ad8ea8e8a14f5f1cebb21a77c6216a9e4ced68b04b5999L17-R18 "Avoid usage of reserved identifiers").

:thought_balloon: How do you think about to avoid any more that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

mintsuki, 2024-01-21:
I am not quite sure what you mean, but beyond the use of reserved identifiers in some places, if there are other instances of invoked undefined behaviour, please let me know.

elfring, 2024-01-21:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?

mintsuki, 2024-01-21:
That's not what I asked. I will also stop replying to you as well, given it is likely either some bot or AI talking, not a real human.

elfring, 2024-01-21:
>That's not what I asked.

:eyes: We know that some update candidates are still remaining.

>I will also stop replying to you as well, given it is likely either some bot or AI talking, not a real human.

Please reconsider such a view.
:crystal_ball: Can I convince you anyhow about further improvement possibilities?

commit 2afb8e64 (2024-01-20): misc: Avoid usage of reserved identifiers in certain cases. Closes #325