Limine-Bootloader/Limine #326: Ability to shut down from limine menu.

amano-kenji, 2024-01-20:
Can limine have a built-in capability to shut down the computer?

With the bloated grub, I could add a menu entry that executes `halt` which shuts down my computer.

Or, is there an EFI executable that shuts down the computer?

PC-Doctor666, 2024-01-22:
> Can limine have a built-in capability to shut down the computer?
> 
> With the bloated grub, I could add a menu entry that executes `halt` which shuts down my computer.
> 
> Or, is there an EFI executable that shuts down the computer? Or, can limine execute an EFI script?

That's funny to think Grub is "bloated".

amano-kenji, 2024-01-22:
grub2 is already difficult to understand. I could not write grub2 config with hand. grub-config generates a config that's not compatible with ZFS root, so I had to insert the correct ZFS kernel parameters in /etc/default/grub. The ZFS bug is still not fixed. It is conceptually bloated. It is buggy.

kovmir, 2024-05-22:
You can press the power button, it does not hurt the computer in any way.

amano-kenji, 2025-03-31:
I think it doesn't hurt to leave it open because my power button is in a awkward position.

phanirithvij, 2025-05-08:
what about a reboot option? like how grub allows
```
menuentry "Reboot" {
  reboot
}
```

amano-kenji, 2025-05-11:
Reboot is okay, but if reboot is available, why can't you implement poweroff?

phanirithvij, 2025-05-11:
I'm not part of the project it was a suggestion as well. leaving this comment to prevent any misunderstanding.

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.

closed 2025-08-31 by qookei