Limine-Bootloader/Limine #330: Onboard to nixpkgs

phip1611, 2024-01-31:
Hi, cool project! 
It would be awesome to upstream this to [nixpkgs](https://github.com/NixOS/nixpkgs)! My company and I do a lot of OS-related development and [Nix](https://nixos.org/). Having everything packaged in Nix enables us to create reproducible and productive development environments.

As this is my last day of work before my long vacation starts, I don't want to upstream this to nixpkgs today. I'm happy to work towards that goal when I'm back in May!

For now, I'd like to share my progress so that anyone else can continue working on this - if interested. Currently, I package it in Nix like that, but without the cross-compilation for riscv64 and aarch64:

```nix
# Builds limine with all available features.

{ gcc, gnumake, mtools, nasm, stdenv }:

let
  # It is recommended by limine to use the release tarballs instead the git repo.
  limineUrl = "https://github.com/limine-bootloader/limine/releases/download/v7.0.2/limine-7.0.2.tar.gz";
  limineSrc = builtins.fetchTarball {
    url = limineUrl;
    sha256 = "sha256:0h70hkzyxpavsa896mj8pqc4gr6hj0bb2d76prp0rsgy1syyy9kc";
  };
in
stdenv.mkDerivation {
  pname = "limine";
  version = "7.0.2";
  src = limineSrc;
  nativeBuildInputs = [ mtools gnumake gcc nasm ];
  doCheck = false;
  # TODO missing --enable-uefi-{aarch64,riscv64} cross compilation so far
  configurePhase = ''
    ./configure \
      --enable-bios-cd \
      --enable-bios-pc \
      --enable-bios-pxe \
      --enable-uefi-ia32 \
      --enable-uefi-x86-64 \
      --enable-uefi-cd
  '';
  buildPhase = ''
    make -j $(nproc)
  '';
  installPhase = ''
    cp -R bin/. $out
  '';
}
```

ping @blitz

TobiasSN, 2024-01-31:
This was already tried at NixOS/nixpkgs#194841, but it was never reviewed by the relevant maintainers

phip1611, 2024-01-31:
Oh, good to know! Yeah, the throughput in nixpkgs is incredible and maintainers are all overwhelmed.

My colleques and I know a few maintainers personally. I'm confident that we can do something about that issue.

TobiasSN, 2024-01-31:
That would be cool, I would love to see Limine in Nixpkgs!

phip1611, 2024-01-31:
From what I've seen in https://github.com/NixOS/nixpkgs/pull/194841:

Maybe it would be good so split that up into two MRs:
- just init the bootloader so that people can use the build artifacts
- make the bootloader usable by NixOS

blitz, 2024-01-31:
Yes, splitting it into smaller individually useful PRs will make it easier to get it into nixpkgs. Adding it as a bootloader option and getting everything right is a lot of (review) work. For that it would be interesting to get at least a few people together that would maintain it. That being said, we can probably help direct some attention to those PRs.

phip1611, 2024-03-18:
Hi, I'm back from vacation and ready to work on that issue now! @TobiasSN

phip1611, 2024-03-18:
Can you please provide some instructions for cross compilation? For example, if I'm on `x86-64-linux` and want to run

```
    ./configure \
      --enable-bios-cd \
      --enable-bios-pxe \
      --enable-bios \
      --enable-uefi-ia32 \
      --enable-uefi-x86-64 \
      --enable-uefi-aarch64 \
      --enable-uefi-cd \
      --enable-uefi-riscv64
```

I'm interested in the build environment for both gcc and clang.

Background: I want (because I _think_ it is the right thing to do) to let the `limine` derivation always build everything. Otherwise, this would limit other cross-compilation toolchains that want to build a bootable image for another platform using `pkgs.limine`.

phip1611, 2024-03-22:
I created a draft MR in nixpkgs. The only blocking thing right now is that:

- one of you has to tell me whether the provided meta information are fine (license)
- you have to tell me whom I should add as maintainer (next to me)
- if you have any other general concerns or so

https://github.com/NixOS/nixpkgs/pull/298048

phip1611, 2024-03-25:
I added @48cf in a dedicated commit, see

https://github.com/NixOS/nixpkgs/pull/298048

phip1611, 2024-03-25:
I just copied the information from https://github.com/NixOS/nixpkgs/pull/194841/commits/b2c893d76838be094612a8710dc4ed1692bd9b6c - sorry!

48cf, 2024-03-25:
whoops, apparently these @users.noreply.github.com email are not usable... can leave that as is then - i hope it's not too much trouble!

phip1611, 2024-03-25:
Got it. Please check the latest version?

48cf, 2024-03-25:
yeah that looks alright, thanks!

phip1611, 2024-04-06:
Closed. Everything done!

closed 2024-04-06 by phip1611

phip1611, 2024-04-07:
General remark. I think, some upstream bots should automatically this in nixpkgs, once a new Github release is created. We can keep an eye on it.

RoadRoller01, 2024-04-30:
nice work guys, so what is the next move to add limine to nixpkgs as option?

side note phip1611 you are man of culture for being persistent and patient

phip1611, 2024-05-01:
> nice work guys, so what is the next move to add limine to nixpkgs as an option?

Thanks! I guess you refer to using Limine as optional bootloader for NixOS.

So, at my company, generally speaking we have interest in Limine. That's why we worked on upstreaming it in our work time at the first place. However, at the moment, we can't justify spending many hours on it with little company benefit. At least I am unsure what the required effort for this is. @astro might have better insights. 

I do not have the capacity to work on this in my free time in the next months, unfortunately.