Limine-Bootloader/Limine #333: Feature Request: Implement grub-style MAC-based PXEBoot limine.cfg configurations.

tfinnegan937, 2024-02-01:
With Grub, when PXEBooting, you can distribute a specific grub.cfg to a specific board depending upon the board's MAC address.

For example, if I had a board with an on-board network interface that had the MAC address FF:FF:FF:FF:FF:FF, you could create the file grub.cfg-01-ff-ff-ff-ff-ff-ff and grub would request that file instead of grub.cfg from the tftp server. (Grub actually just goes down a list of alternatives to grub.cfg, and only grabs grub.cfg if none of the others exist on the server).

It would be extremely useful for limine to implement this feature with limine.cfg. Grub has the unfortunate flaw that it uses a maximum TFTP block size of 1024 on both UEFI and BIOS systems...It's quite slow. Limine makes much more efficient use of the network when requesting files from the TFTP server than grub does.

My particular usecase is that I have a lab environment where we have a rolling rack that contains several single-board computers. We compile unique RTOS kernel images for each board, and when we want to update the image, we replace the file on our tftp server and reboot the board. 

This page from the GRUB manual describes the feature that I am referring to: https://www.gnu.org/software/grub/manual/grub/html_node/Network.html

blanham, 2024-03-05:
I don't know about your specific situation, but I once was thinking about such a feature, before discovering that dnsmasq, which I use as a DHCP proxy for serving PXE requests, already has the ability to select payloads based on machine MAC address (among other filtering options). Would this work for your setup?

mintsuki, 2024-03-09:
If that's the case I'll close as not planned.

closed 2024-03-09 by mintsuki