Limine-Bootloader/Limine #334: Migrating to QOI?

omasanori, 2024-02-06:
It may be a naive idea, but [QOI](https://qoiformat.org/) sounds suitable to Limine, as it [performs well](https://qoiformat.org/benchmark/) and its simplicity is desirable to avoid certain bugs like LogoFAIL. Replacing stb_image with a QOI decoder would reduce the size and complexity of the bootloader without sacrificing cool boot screens.

mintsuki, 2024-02-06:
Hi and thanks for opening an issue.

For one, replacing all supported image formats with QOI would be a big change, which breaks backwards compatibility and thus requires a new major version, so it wouldn't be very practical.

That said, I oppose dropping support for the more common image formats in favour of the relatively obscure QOI format; that means that adding QOI support would be something additional, next to the already supported image formats, thus making the "simplicity" argument irrelevant...

Now, even assuming stbi has an issue that would allow a similar exploit to LogoFAIL, correct if I am wrong, that would involve the attacker being able to replace the image to be used at will... this is actually easily preventable by enforcing the blake2b hash of the image file intended to be loaded.

So yeah, while I am not necessarily opposed to adding support for QOI in addition to the rest, we're not migrating to solely QOI anytime soon.

omasanori, 2024-02-06:
Thank you for describing the design rationale!