Limine-Bootloader/Limine #338: Bug: Limine does not support being loaded above 4GiB

ethindp, 2024-03-19:
I encountered this bug when trying to use Limine as a boot loader on my actual machine fr my Arch Linux install (before I switched to just using efistub). I find this lack of support a bit surprising, since I can't reproduce it in Qemu when using OVMF, and I'm using TianoCore with Coreboot on my actual machine which is very similar to Qemu's OVMF, I think. But the lack of support is surprising since I've never seen a boot loader care about where it's loaded, and I would think that if it's compiled for UEFI, it shouldn't care where the firmware happens to load it, since where the image is loaded is left up to the implementation. Is this intentional?

mintsuki, 2024-03-19:
The error is intended, on x86. Please refer to https://github.com/limine-bootloader/limine/issues/206 for the rationale behind that.

Now, I was under the impression that x86 systems pretty much never loaded EFI applications above 4GiB, which is why I took the simpler approach to gracefully fail if there so happens to be a system that does load above 4GiB.

Is this machine special in some way? You mentioned Coreboot... that indeed does sound like a non orthodox setup. I will nonetheless now have to add the ability for Limine to work when loaded above 4GiB one way or another.

ethindp, 2024-03-19:
I mean, you don't *have* to add it if you believe it's simpler. I'm uncertain how adding it complicates things (I'm well-versed in x86 but no expert) but it was something I'd never seen before. The machine is a System76 Lemur Pro 10 running System76's Coreboot 2023-10-13_5d997d1, 10/12/2023. Frankly I was surprised that it had loaded it above 4GiB; like you I didn't think firmware did that, but... Apparently some do. I know the UEFI spec is vague about where images can/should be loaded (and for good reason) but I figured most UEFI implementations were 32-bit so loading above 4GiB was typically not a problem. But maybe this firmware initializes the processor directly to 64-bit mode? It's a feature in Coreboot but last time I checked it was still marked as experimental...

mintsuki, 2024-03-25:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/14748488/BOOTX64.EFI.gz)

Can you please try this image to see if it works on your system? (Remove the `.gz` without uncompressing it, it's just a dummy extension so GitHub swallows it)

mintsuki, 2024-03-26:
( @ethindp )

mintsuki, 2024-03-28:
In 7.2.1, the workaround I pushed (and sent above for you to test) is included. This should mitigate the issue in most instances where it is possible to do so. If the issue is not fixed for you, let me know by reopening the issue and/or adding comments here.

closed 2024-03-28 by mintsuki