Limine-Bootloader/Limine #346: Make limine compatible with the boot loader specification by supporting an XBOOTLDR partition

emk2203, 2024-04-18:
[There's a bootloader specification](https://uapi-group.org/specifications/specs/boot_loader_specification/) by the [UAPI group](https://github.com/uapi-group) (systemd, Meta, SUSE, Redhat, Microsoft, Edgeless Systems, NixOS, Canonical, Amazon, Free Desktop, Gnome, carbonOS) which emerges as the new industry standard.

This specification calls for support of an extended boot loader partition (XBOOTLDR) in addition to the ESP partition if present.

Could limine support this XBOOTLDR partition?

This would require to use the XBOOTLDR partition if it is present instead of the ESP partition (see linked specification).

Boot partition would be:
>    On disks with an MBR partition table: → the boot partition (no change)
>    On disks with a GPT partition table: → the XBOOTLDR partition if it exists
>    Otherwise, on disks with a GPT partition table: → the ESP

mintsuki, 2024-04-18:
No, sorry. Limine will not conform to that specification in this, or any other aspects.

closed 2024-04-18 by mintsuki

sylv256, 2024-07-08:
> > No, sorry. Limine will not conform to that specification in this, or any other aspects.
> 
> Why not support the standard? This limits it's utility, and separating and clearly defining boot and the ESP seems quite reasonable.

This is a simplistic project, and spamming the issue tracker with nonsense like this demanding that open-source developers extend their free labor to your niche usecases is rude and disrespectful.

Kangie, 2024-07-08:
> This is a simplistic project

That's nice. That doesn't mean that we shouldn't document the reasons _why_ a decision not to implement an open standard related to the core purposes of the product was made.

> spamming the issue tracker with nonsense like this

This isn't your project, why are _you_ so salty?

> demanding that open-source developers extend their free labor to your niche usecases is rude and disrespectful.

I don't see any demands here, just several polite requests.

Please take your outrage on behalf of FLOSS developers somewhere else and let us try and be productive here. Spamming the issue tracker with nonsense like this, demanding that users avoid asking questions for fear of receiving harassment, is rude and disrespectful.

I was looking into the bootloader after several of my distributions's users had mentioned it recently and it's in our unofficial repository which may lead to it being picked up for the main repos.

As a distro developer I am interested in the rationale for a system that chooses to NIH a standard. This interaction has soured me on the project.

mintsuki, 2024-07-09:
> As a distro developer I am interested in the rationale for a system that chooses to NIH a standard. This interaction has soured me on the project.

I don't see why an interaction with someone which, as you put it, is not a maintainer nor developer of the project should sour you on the project itself.

I believe my initial refusal to consider this was pretty polite, but admittedly I did not expand on the reasons for my refusal, so let me do so now.

First of all, the XBOOTLDR partition. There is not *need* for Limine to "support" this as one can use the PARTUUID of an XBOOTLDR partition in the Limine config to access files on it. Unless what is meant is that Limine should find the `limine.cfg` file on an XBOOTLDR partition instead of an EFI partition or any other partition first. While this is possible and doable, I don't see the point in implementing such a subset of this specification, which is evidently much more expansive than just talking about XBOOTLDR, as it also dictates where to find boot entries, how they should be formatted, etc, etc.

As to why Limine shouldn't comply with *all* of this specification. Well, this specification is one of those very Linux-centric things that puts very little emphasis on other, non-Linux, OSes, which Limine also caters to. It would also be quite an extensive compatibility break, which is not desirable, else both the current boot menu method and this specification should be handled simultaneously which would be messy.

And my final, personal, note about this specification is simply that I do not like it and I do not like how it would restrict and/or change my vision for how Limine (or any bootloader really) should operate and behave. Therefore I will, politely, refuse to adhere by it.

Now, I am sorry to hear that my refusal may limit the usefulness of Limine, but I am also not changing my take on the project, which by the way has pretty much existed for half a decade now, in order to serve the users of an unspecified Linux distro. But if I may make a suggestion, I feel like it would be easy to compile, at runtime/kernel install time, a `limine.cfg` file off of the bootloader specification files, which would make this whole discussion pointless to begin with, as it would be a win-win for everybody.

Kind regards.

P.S.: Please do not get mad at someone unaffiliated with the project that simply wanted to defend me (or any other Limine developer) on our behalf. I am perfectly capable of defending myself or my opinions without the need of a third party, so if you're gonna be mad at anyone, please let that be myself.

ElectrodeYT, 2024-07-09:
I also would like to note one thing I instantly noticed about XBOOTLDR:

> Optionally, an Extended Boot Loader Partition (XBOOTLDR partition for short) — a partition with GPT type GUID of bc13c2ff-59e6-4262-a352-b275fd6f7172 — may be used as the second of two locations for boot loader menu entries. This partition must be located on the same disk as the ESP.

The word "optionally" here would indicate to me that supporting this parition is not required to conform with the specification.

Kangie, 2024-07-09:
> I don't see why an interaction with someone which, as you put it, is not a maintainer nor developer of the project should sour you on the project itself.

As an outsider I find it quite reasonable that, with this event being my first interaction with your community, something like this would taint my opinion of the project; I certainly don't see anybody discouraging this behaviour.

mintsuki, 2024-07-09:
> As an outsider I find it quite reasonable that, with this event being my first interaction with your community, something like this would taint my opinion of the project; I certainly don't see [...]

Well, first off, you're acting like what @sylv256 said was completely unreasonable and mean, which it certainly did not seem to be, to me.

Second, sylv256's comment was made 9 hours ago; most of the time between then and now I was sleeping. Do you think you're being reasonable in the slightest to think I should moderate comments on dead issues made while I was sleeping, which aren't even that mean or problematic anyways? If you feel attacked so badly to need prompt intervention, report the comment to GitHub; they are quite fast in intervening when it comes to unjust interactions.

In general, it seems to me, and I hope that I am wrong, that these "this interaction taints my opinion of the project" or "this interaction has soured me on the project" statements are cheap psychological tricks to get me to bend over to your will, whichever that may be, and that, I do not like. I repeat, though, that I hope that I am wrong in reading your statements that way. In any case know that that won't work on me.

Kangie, 2024-07-09:
> As to why Limine shouldn't comply with all of this specification. Well, this specification is one of those very Linux-centric things that puts very little emphasis on other, non-Linux, OSes, which [...]

Thanks for elaborating on your rationale for not accepting this request. It seems reasonable and while I still think supporting the BLS would be ideal I respect your right to direct the project as you see fit.

Good luck with your community.

mintsuki, 2024-07-09:
> > As to why Limine shouldn't comply with all of this specification. Well, this specification is one of those very Linux-centric things that puts very little emphasis on other, non-Linux, OSes, which Limine also caters to. It would also be quite an extensive compatibility break, which is not desirable, else both the current boot menu method and this specification should be handled simultaneously which would be messy.
> 
> Thanks for elaborating on your rationale for not accepting this request. It seems reasonable and while I still think supporting the BLS would be ideal I respect your right to direct the project as you see fit.
> 
> My objection was to being accused of spam for requesting additional information, not you rejecting the request which is your right of course.
> 
> Good luck with your community.

Thanks for understanding. I don't mean there to be any bad blood between me or my project and you or the Gentoo community. What that person said was rather rude, on second read; I will mark it as off-topic and hide it, and I am sorry it bothered you. At first it seemed to me like just some generic commentary on people annoyingly requesting features of open source projects, rather than some personal attack on *you*, but I guess it could be interpreted that way.

In any case I'll repeat that they are not affiliated with the project in any way, and it wouldn't be any different than thinking that, e.g., the Meson project condones some random opinion or rude statement if I leave such a statement in their issue tracker, and for whatever reason no one really does anything about it. Especially considering that it is mostly just me here, and not a whole team of moderators.

Cheers.