Limine-Bootloader/Limine #347: Config file not found

kovmir, 2024-04-21:
### Short description

I can boot into Limine version 7.4 bootloader, but Limine is unable to see its config file, so it says. I tried all the [locations listed in the documentation](https://github.com/limine-bootloader/limine/blob/d0bbe19358d980b400d626c9959c5c80854689cf/CONFIG.md?plain=1#L5). I cannot reproduce it on any other devices of mine.

### Details

I am on [Thinkpad x230](https://www.thinkwiki.org/wiki/Category:X230), [Slackware Linux](http://www.slackware.com/) 15.0. `limine.cfg` is located at the root of the UEFI partition:

```
$ findmnt /boot/efi
TARGET    SOURCE    FSTYPE OPTIONS
/boot/efi /dev/sda1 vfat   rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro
```

```
$ ls -l /boot/efi
total 3
drwxr-xr-x 5 root root 1024 Apr 21 18:12 EFI
drwxr-xr-x 2 root root 1024 Mar 13  2023 limine
-rwxr-xr-x 1 root root  340 Apr 20 17:31 limine.cfg
```

I have only one drive in this computer:

```
$ lsblk -d
NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda    8:0    0 223.6G  0 disk
```

GitBassador, 2024-04-21:
You need to make and provide your config file with stuff in it.
Depending on what its contents are, you want it wherever your linux images are.
It must be on a fat32 partition, and somewhere other than sda1.
I assume your possibly dualbooting with windows.

mintsuki, 2024-04-21:
@kovmir Hello,

How does Limine 7.4.0 say that it cannot find the config? Asking because since 7.4.0 specifically, Limine won't say anything in particular if there is no config/config has no entries, and it will instead just present an empty main menu, at least in theory.

kovmir, 2024-04-21:
> How does Limine 7.4.0 say that it cannot find the config? Asking because since 7.4.0 specifically

Limine 7.3 displays a message saying the config has not been found occupying the entire screen width. Limine 7.4 presents an empty menu just like you have described.

mintsuki, 2024-04-21:
> > How does Limine 7.4.0 say that it cannot find the config? Asking because since 7.4.0 specifically
> 
> Limine 7.3 displays a message saying the config has not been found occupying the entire screen width. Limine 7.4 presents an empty menu just like you have described.

Has this worked on a previous Limine release? If so, which?

Also please triple check that you indeed do have that `limine.cfg` file in the right partition.

kovmir, 2024-04-21:
> You need to make and provide your config file with stuff in it.

I have provided the config file for Limine. Do you mean I should post the contents of it here? OK:

```
TIMEOUT=5

:Slackware
        PROTOCOL=linux
        KERNEL_PATH=boot:///EFI/Slackware/vmlinuz-generic
        MODULE_PATH=boot:///EFI/Slackware/initrd.gz
        CMDLINE=root=/dev/sda2 vga=normal ro
```

> Depending on what its contents are, you want it wherever your linux images are.

There is no such thing in the documentation.

> It must be on a fat32 partition, 

It is.

> and somewhere other than sda1.

No such thing in the documentation.

> I assume your possibly dualbooting with windows.

No, I do not. What made you assume that?

> Edit: its probably loooking for it in the folder named limine

[At first, it is looking for the config file at the root of the partition](https://github.com/limine-bootloader/limine/blob/d0bbe19358d980b400d626c9959c5c80854689cf/CONFIG.md?plain=1#L8).

kovmir, 2024-04-21:
> Has this worked on a previous Limine release? If so, which?

No it has never worked, I tried Limine version 4 before you switched over to semantic versioning. I did not bother reporting at the time.

> Also please triple check that you indeed do have that `limine.cfg` file in the right partition.

My first message shows the config file is indeed at root of the EFI partition, and `sda` is my only drive. You can also look at the entire partition:

```
$ tree /boot/efi
/boot/efi
├── EFI
│   ├── BOOT
│   │   ├── BOOTX64.EFI
│   │   └── limine.cfg
│   └── Slackware
│       ├── elilo.conf
│       ├── elilo.efi
│       ├── initrd.gz
│       ├── vmlinuz-generic
│       └── vmlinuz-huge
├── limine
│   ├── limine.cfg
│   └── liminex64.efi
└── limine.cfg

4 directories, 10 files
```

As you can see, I tried putting `limine.cfg` in several places.

mintsuki, 2024-04-21:
Is there a chance you could image `/dev/sda1` as is and send it over? That would be very useful.

kovmir, 2024-04-22:
> Is there a chance you could image `/dev/sda1` as is and send it over? That would be very useful.

[Here](https://dl.kovmir.cc/downloadFile?id=Vbmo09f5iIJ4V7w) (expires in 20 days).

mintsuki, 2024-04-22:
> > Is there a chance you could image `/dev/sda1` as is and send it over? That would be very useful.
> 
> [Here](https://dl.kovmir.cc/downloadFile?id=Vbmo09f5iIJ4V7w) (expires in 20 days).

Alright, so...
it seems like your FAT filesystem was generated by very ancient tools; it lacks a proper filesystem type field in its BPB (aka superblock) which Limine needs in order to identify the filesystem.

No worries, this can be fixed by running:
```bash
echo "FAT     " | dd conv=notrunc bs=1 count=8 of=/dev/sda1 seek=82
```
(as root, and assuming `/dev/sda1` is still the partition in question)

PS: I see that you have comments in the config in the form of
```ini
SOME=string # comment
```
This is not a valid comment. Comments can only start at the beginning of a line until the end of the line.

kovmir, 2024-04-22:
It works, thank you.

closed 2024-04-22 by kovmir