Limine-Bootloader/Limine #350: Implement loopback for booting disk image files

austanss, 2024-05-12:
It would be useful and beneficial to allow a user to load disk image files from the local filesystem with a loop-back configuration of some sort. This would be useful to a Linux user who wants to install a distribution that releases in ISO formats, and does not have a dedicated device at hand to write the image to. The grub boot-loader also has an option to setup loop-back images and load EFI executable files from them. This could also be added as a feature without introducing any breaking changes or significant design shifts. This may require a non-trivial effort of code to implement, but the effort does not seem insurmountable in my own view.

**Example configuration:**
```
TIMEOUT=3

:Debian Installer
    PROTOCOL=loopback-iso
    IMAGE_PATH=boot:///debian.iso
    
```
I will take effort to research the codebase on my own merit to investigate the possibilities for these options.

austanss, 2024-05-13:
I've gained a little bit of a grip in this area of the code, and I am experimenting with some changes...

mintsuki, 2024-05-13:
The problem, and the reason why this has never been done, is not getting the concept of a loopback device going, but rather the fact that this is not possible to get working without having kernels that cooperate (either by native support, or, like Ventoy does for example, by shipping kernel modules for each individual OS kernel) to support these loopback devices for what is needed to finish booting and beyond.

austanss, 2024-05-13:
In that context, then, I'll limit myself in scope, for now, to investigating loopback options specifically for booting Linux distributions.

austanss, 2024-07-02:
If I'm being honest, I haven't made progress towards this in a while and it's unlikely that I will actually make more any time soon. This is in large part because the original personal use case I found myself in is no longer applicable (because I went and bought a flash drive) in my current circumstances.

While it's obviously still possible, and I'd stand by that, I'm not really in the position to implement this myself...

closed 2024-08-04 by mintsuki