Limine-Bootloader/Limine #359: Black screen if not using TEXTMODE

HirbodBehnam, 2024-05-29:
Hi
I was tampering with limine and I was wondering if you guys can help me find where I'm making a mistake. I'm trying to build a simple compiled Linux kernel + Busybox combo and use limine as its bootloader. However, if I do not enable `TEXTMODE=yes`, I'm not able to see anything after the boot menu.

Because I'm compiling the Linux kernel myself, I'm speculating that I'm miss-configuring something in the kernel config. Probably related to frame buffers? I enabled framebuffers but it didn't help. I al
so tried booting in BIOS and UEFI with QEMU and OVMF but it didn't help. I'm also building an ISO file using the documentation of limine.

I'm attaching my configuration to this message and I appreciate any help. I've also attached a simple ISO I've built.

[config.zip](https://github.com/limine-bootloader/limine/files/15489658/config.zip)
[image.zip](https://github.com/limine-bootloader/limine/files/15489663/image.zip)

mintsuki, 2024-06-04:
Hi, sorry for the late reply!

This is very likely a Linux configuration issue; I do not remember off the top of my head what config option is the one enabling or disabling graphical terminal output, sorry. Please do let know if you find out!

closed 2024-06-04 by mintsuki

HirbodBehnam, 2024-08-11:
Hi
I had this one in my backlog for a long time and today I had the time to investigate this. As I was expecting, this had to do something with the framebuffer. But the problem was that I actually missed the fact that the "Support for frame buffer device drivers" has a submenu which you can select drivers and in that menu, you should select the corresponding driver or "EFI Based framebuffer" if you are booting in UEFI mode. If probably need VGA drivers if you are booting in BIOS.

Here is my example config file:
[config.zip](https://github.com/user-attachments/files/16573994/config.zip)

Cheers and thanks for the great project!