Limine-Bootloader/Limine #380: RISC-V port does not respect MAX_PAGING_MODE=Sv39

robotman2412, 2024-07-06:
Despite this config:
```
TIMEOUT=0

:BadgerOS
    PROTOCOL=limine
    KASLR=no
    KERNEL_PATH=boot:///boot/badger-os.elf
    MAX_PAGING_MODE=Sv39
```
Limine loads BadgerOS with the Sv48 paging mode. BadgerOS supports this but my real hardware does not. To verify my kernel in QEMU, I would like to lock Limine to Sv39 even on systems that support larger paging modes.

48cf, 2024-07-06:
MAX_PAGING_MODE is not something you can change through the config. Use the [Paging Mode Request](https://github.com/limine-bootloader/limine/blob/v7.x/PROTOCOL.md#paging-mode-feature) instead.

robotman2412, 2024-07-06:
> MAX_PAGING_MODE is not something you can change through the config. Use the [Paging Mode Request](https://github.com/limine-bootloader/limine/blob/v7.x/PROTOCOL.md#paging-mode-feature) instead.

In that case, a note should be made about MAX_PAGING_MODE, because `CONFIG.md` states the contrary.

48cf, 2024-07-06:
It doesn't..? Where do you see that?

48cf, 2024-07-06:
oh wow my bad, yeah i didn't know that

48cf, 2024-07-06:
It looks like you still need the Limine request for the config option to take effect, but I might be misreading the source code.

closed 2024-07-06 by mintsuki

robotman2412, 2024-07-06:
Fixed same-day, nice

commit 50a949dd (2024-07-06): protos/limine: Do not ignore MAX_PAGING_MODE config setting if paging mode request missing. Closes #380
commit 1e60942e (2024-07-06): protos/limine: Do not ignore MAX_PAGING_MODE config setting if paging mode request missing. Closes #380