Limine-Bootloader/Limine #382: Feature Idea "Boottime Linker"

Shinribo, 2024-07-10:
I had a short discussion on the OSDev Discord with the resulting idea of Boottime linking done by a bootloader. After thinking about it for a while, I came to the conclusion that the idea might actually be useful as it would reduce the complexity of the boot process and increase modularity of a kernel (for example the boot driver can be different for each install, acpi and DT-Driver dont need to be statically linked and a kernel can use a plattform optimized hal).

untodesu, 2025-01-06:
I think that can be achieved with an existing feature set: you load desired modules as bootloader modules and then use bootloader API tags to figure out where each one resides; if modules are treated as data, setting up a VMM and re-mapping them as executable would solve that as well

marv7000, 2025-01-06:
IMO this feature would go against Limine's philosophy. This is unneeded complexity for something that should be handled by a kernel itself.

mintsuki, 2025-03-03:
Sorry, the feature is not planned.

closed 2025-03-03 by mintsuki