Limine-Bootloader/Limine #386: v5 cannot enable A20 line

bkauler, 2024-07-20:
I know that v5 is no longer supported, but thought I would ask just in case this problem has occurred in a later release and maybe I can backport the fix.

We are stuck on v5 as require ext4 support. 

Puppy Forum member 'tosim' reported getting the error "Cannot enable A20 line".
He has an older computer, HP5800.

I think read that there are different ways of enabling the A20 line. I'm wondering if there has been any modifications to enabling A20 since 5.20231207.1 was released. A long-shot, I know.

mintsuki, 2024-07-20:
Hi, these 2 commits may fix the issue.

Additionally, I went ahead and reverted the commit where ext2/3/4 support was dropped, which means it will be back in 7.12.0, albeit with a notice that it is experimental and unmaintained/unsupported upstream. I hope this will do! (Reopen if necessary)

closed 2024-07-20 by mintsuki

bkauler, 2024-07-21:
From our point of view, EasyOS, this is very great news! Thank you so much!

I understand the ideology of wanting to keep everything simple and not be bothered with supporting filesystems other than fat.

bkauler, 2024-07-21:
I have posted about the return of ext4 in the Puppy Forum:

https://forum.puppylinux.com/viewtopic.php?t=12214

mintsuki, 2024-07-21:
Nice! Well, 7.12.0 has been released, enjoy. (Also please let me know about that A20 issue, if it's still there)

commit b3385b80 (2024-07-20): bios/a20: Improve a20_check(). Could fix #386
commit 63427dca (2024-07-20): bios/a20: Add support for fast A20 method. May fix #386
commit b6da3ed8 (2024-07-21): bios/a20: Improve a20_check(). Could fix #386
commit f3281459 (2024-07-21): bios/a20: Add support for fast A20 method. May fix #386