Limine-Bootloader/Limine #399: Describe the stack alignment or ABI required on for the entry function

adsnaider, 2024-08-10:
As far as I can see, the limine protocol specification doesn't actually specify if the stack will be aligned to 16-bytes on entry, or offset by 8 bytes. If the entry is required to follow the `C` ABI, then that would be sufficient. Experimentally, the stack is at an 8-byte offset alignment which matches the C ABI. This is actually contrary to `_start` function which assumes a 16-byte aligned stack. I don't really care which it is I mostly want to make sure it's well-defined and documented somewhere.

AshesOfEther, 2024-08-10:
This is specified in the "General Notes" section:

> The calling convention matches the C ABI for the specific architecture (SysV for x86, AAPCS LP64 for aarch64, LP64 for riscv64).

mintsuki, 2024-08-10:
Yeah, it's described there. Closing the issue, if you have any other concern please let me know.

closed 2024-08-10 by mintsuki