Limine-Bootloader/Limine #402: Limine request issues

ProfessorLongBeard, 2024-08-13:
After getting started with the Limine bare-bones documentation, and getting a test kernel started Limine seems to throw an error about lower half kernels not being supported despite following the linker script example for a higher half kernel. This seems to only happen when performing requests from Limine. An example would be when I would request for a 4-level (minimal) paging setup from limine. Mind you I followed the provided documentation 1 to 1.

mintsuki, 2024-08-13:
Please post a reproducible example or link to code. Possibly as minimal as you can just to reproduce the issue claimed.

ProfessorLongBeard, 2024-08-13:
> Please post a reproducible example or link to code. Possibly as minimal as you can just to reproduce the issue claimed.

As soon as I can I will since I've got to head to work

closed 2024-08-13 by ProfessorLongBeard