Limine-Bootloader/Limine #404: Question: How to print in limine based c/rust kernels to screen? Is there any example implementation of flanterm?

commitcompanion, 2024-08-22:
(Originaly asked on the flanterm repo, also open there)
[main.rs.txt](https://github.com/user-attachments/files/16647735/main.rs.txt)
How do you print text in limine-rust-template using flanterm_bindings crate? Attached is an attempt to do this.
info: first asked in an issue in rust limine template, seemed to be the wrong place to ask according to lylythechosenone. Many thanks for an answer!

commitcompanion, 2024-08-25:
well, maybe this project helps:https://github.com/ilobilo/limine-terminal

mintsuki, 2024-08-25:
Closing this as off topic. Drawing text to the screen is not a Limine related thing. You may want to ask whoever maintains the Flanterm bindings crate.

closed 2024-08-25 by mintsuki