Limine-Bootloader/Limine #417: Limine doesn't load on Pure UEFI (no CSM support) on a ThinkPad X240.

AFellowSpeedrunner, 2024-10-12:
I'm not quite sure if Limine is supposed to work with CSM support enabled only but I've observed my ThinkPad not getting to the Limine menu on Pure UEFI and staying at a black screen after the ThinkPad logo.

On UEFI with CSM support enabled, it gets to the Limine menu just fine. I'm not quite sure what is causing this issue or if this is just a hardware specific thing but I thought I should make an issue nonetheless.

mintsuki, 2024-10-13:
Thanks for opening this issue.

Just to clarify, when you say CSM support enabled/disabled, do you mean actually using the legacy BIOS support given by CSM or still using UEFI but with CSM enabled?

Limine should work on this machine, with or without CSM enabled. But depending on what you mean, it could be that you simply did not install the Limine variant for UEFI (or did not install it correctly), while only having the BIOS variant installed.

AFellowSpeedrunner, 2024-10-13:
No problem!

What I mean by CSM is the UEFI with CSM enabled.

These images should show what I mean.

[image]

[image]

When CSM is set to No (Pure UEFI without any Legacy components) Limine doesn't load.

I also installed Limine with pacman when I installed Arch.

I hope this clears up the confusion.

mintsuki, 2024-10-13:
Is this issue specific to Limine only? Can it not be reproduced with any other bootloader?

As far as I can tell you, Limine should not care whether CSM is on or off if booting the Limine UEFI port. Can you confirm Limine is even entered at all before the system hangs?

What was the reason you attempted to switch CSM off in the first place? Can you double check that "UEFI only" with CSM on actually means the system truly never attempts to boot with CSM? (Disregard what the firmware setup help text says, firmware is often broken. Just boot with CSM on and "UEFI only" and double check you are actually using UEFI).

AFellowSpeedrunner, 2024-10-13:
I think its specific to Limine. I tried with GRUB and systemd-boot UEFI without CSM. Closest result is that GRUB gave me no output on screen but still continued running in the background and booted in.

Limine doesn't enter at all. It just hangs at the black screen with the Lenovo logo.

I wanted everything to use UEFI with no CSM. With UEFI and CSM on, the system switches screen resolutions reverting to a more old BIOS look and resolution (but I changed that to my screens max resolution in Limine so it switches back after the logo).

UEFI with CSM set to no looks like this on boot:
[image]

UEFI with CSM set to yes looks like this on boot:
[image]

Booting with Legacy gives the same screen as UEFI with CSM set to yes, but the system will never boot Limine as it is setup for UEFI.

mintsuki, 2024-10-13:
Sounds like with CSM off the UEFI doesn't provide a linear framebuffer. Could you try adding `serial: yes` to the Limine config?

AFellowSpeedrunner, 2024-10-13:
> Sounds like with CSM off the UEFI doesn't provide a linear framebuffer. Could you try adding `serial: yes` to the Limine config?

I added "serial: yes" to my config. System still hangs after turning off CSM and rebooting to Limine. With CSM on though, my resolution settings are now ignored with it added in.

mintsuki, 2024-10-13:
Yeah... that's expected. Though the situation overall doesn't sound right at all, as if something's seriously wrong with this machine's firmware.

Would it be too much to ask to see if there is any BIOS update available and if so, to install it and try again?

AFellowSpeedrunner, 2024-10-13:
> Yeah... that's expected. Though the situation overall doesn't sound right at all, as if something's seriously wrong with this machine's firmware.
> 
> Would it be too much to ask to see if there is any BIOS update available and if so, to install it and try again?

I'll check. The machine is currently on version GIET67WW 2.17 from 2014-01-08.

AFellowSpeedrunner, 2024-10-13:
I updated the machine to GIET99WW 2.49 from 2020-03-17. With CSM disabled, the same issue occurs. I'm not sure this is a machine problem then considering the BIOS was reflashed and the Embedded Controller was also reflashed.

mintsuki, 2024-10-13:
I'll look into it and let you know. In the meantime I suggest simply keeping CSM on when using Limine on the machine.

AFellowSpeedrunner, 2024-10-14:
I have upgraded the ThinkPad x240 to a 1TB SSD from my old 512GB HDD. I can confirm this is not a machine issue nor a drive issue.

closed 2024-10-15 by mintsuki

AFellowSpeedrunner, 2024-10-15:
Thanks for making a fix commit in v8.1.2. I'm currently waiting for it to enter the Arch repos and hopefully it should go in soon. I'll let you know how it goes. :)

commit 163647f1 (2024-10-15): mm/pmm: Ignore UEFI memory map entries of length 0. Fixes #417