Limine-Bootloader/Limine #422: Feature request: Provide built-in macro for firmware detection

kouosi, 2024-10-23:
Currently there is no built-in method to determine the firmware type (e.g., UEFI or BIOS) within configuration files, specifically for `interface_branding`. If I'm mistaken, please correct me.

This allow us to show menu interface_branding as that of limine default which include firmware like
```config
interface_branding: Void linux (${ARCH}, ${FW_TYPE})
```
It would also be useful in other contexts, such as adding the firmware type to the kernel command line
``` config
kernel_cmdline: -fw_type ${FW_TYPE}
```
So a built-in macro should be introduced, similar to the existing `ARCH` macro. This new macro could be named `FW_TYPE` or `FIRMWARE`, which provides firmware type like UEFI, BIOS etc.

AshesOfEther, 2024-10-23:
You can already detect the firmware type in your kernel using the [Firmware Type feature](https://github.com/limine-bootloader/limine/blob/v8.x/PROTOCOL.md#firmware-type-feature).

kouosi, 2024-10-24:
So is there any way to solve this `interface_branding: Void linux (${ARCH}, ${FW_TYPE})`?

mintsuki, 2024-10-24:
> So is there any way to solve this `interface_branding: Void linux (${ARCH}, ${FW_TYPE})`?

Not at the moment. But I do like the idea and I will implement it soon, thanks!

commit 29907c9d (2024-10-27): config: Add and document FW_TYPE built-in macro. Fixes #422
closed 2024-10-27 by mintsuki