Limine-Bootloader/Limine #425: How do I  solve limine.conf issue?

baponkar, 2024-10-29:
I have `limine.conf` inside of `iso_boot\limine\limine.conf` along with `limine-bios-cd.bin` ,  `limine-uefi-cd.bin` and `limine-bios.sys` file .The `limine.conf` file is look like:
```bash

# Timeout in seconds that Limine will use before automatically booting.
timeout: 60

graphics: yes

interface_resolution: 1366x768

wallpaper: /boot/a.bmp
wallpaper_style: centered

interface_branding: KeblaOS v1.0
interface_branding_colour: 4

# The entry name that will be displayed in the boot menu.
/KeblaOS
    # We use the Limine boot protocol.
    protocol: limine
    

    # Path to the kernel to boot. boot():/ represents the partition on which limine.conf is located.
    kernel_path: boot():/boot/kernel.bin
```

The output look like 
[image]

Not showing Wallpaper or set proper resolution

baponkar, 2024-10-29:
I have found solution `wallpaper: boot():/boot/a.bmp`

baponkar, 2024-10-29:
But resolution issue not solved

closed 2024-10-29 by baponkar

mintsuki, 2024-10-29:
Hi,

The resolution needs to be supported by the machine, which is not a given. Have you tried some other resolution? In any case, you should take it as a hint, given you cannot know in advance whether a machine supports a certain resolution or not.