Limine-Bootloader/Limine #426: Recovery screen when trying to chainload Windows on another drive

adurmk1, 2024-11-01:
Hi all,
i've run into an issue when trying to chainload Windows on a different drive from the drive limine is on.

Here is my working config:
```
timeout: 5
quiet: no
remember_last_entry: yes
interface_resolution: 2560x1440
interface_branding: Limine Bootloader
term_palette: 282c34;ff6c6b;98be65;ecbe7b;51afef;c678dd;5699af;dfdfdf
term_background: ff282c34
term_foreground: dfdfdf
editor_enabled: yes
editor_highlighting: yes
editor_validation: yes

/Arch Linux
    comment: Arch Linux system for productivity
    protocol: linux
    kernel_path: boot():/vmlinuz-linux
    cmdline: root=UUID=2dee1964-e18f-4d97-8c91-d7ee6e58872b
    module_path: boot():/amd-ucode.img
    module_path: boot():/initramfs-linux.img

/Arch Linux Gaming
    comment: Arch Linux system for gaming
    protocol: linux
    kernel_path: guid(2300b56b-8a70-4967-a09c-45cc60dfa95e):/vmlinuz-linux
    cmdline: root=UUID=bb39606e-938c-4aea-adc2-fdee7e4ba668
    module_path: guid(2300b56b-8a70-4967-a09c-45cc60dfa95e):/amd-ucode.img
    module_path: guid(2300b56b-8a70-4967-a09c-45cc60dfa95e):/initramfs-linux.img

/Windows
    comment: Windows (blergh)
    protocol: efi_chainload
    image_path: boot():/EFI/GRUB Windows Chainloader/grubx64.efi
```
Here the windows boot works by chainloading grub with timout 0, that then chainloads windows.
But when trying to chainload windows directly by changing the last line to
```
image_path: guid(9cf95c4d-6a9e-43f6-9e08-3a4a1fc2a312):/EFI/Microsoft/Boot/bootmgfw.efi
```
The Windows Boot Manager seems to successfully get loaded but instantly bluescreens and tells me that my windows installation is broken and to recover windows. The issue disappears when doing the janky workaround with grub.

Here is the blue screen.
[image]

mintsuki, 2024-11-07:
Hi, sorry for the late reply, but I've been unable to reproduce the issue locally.

adurmk1, 2024-11-07:
Ok thx, could be some local setting or something, just wanted to ask if it was something wrong with limine but it seems to be some Windows BS. Anyways I've already found the janky GRUB workaround for my system for the one time every half a year I have to use windows.

closed 2024-11-07 by adurmk1

mieboo, 2026-06-11:
> > Unfortunately no, the workaround I used was removing the Windows entry from limine since it didn't work anyway and remembering the boot order key for the UEFI to directly boot into Windows. Also if you look at my config in the original post, I created a custom grub config which instantly chainloaded windows, so you could install grub just to boot into windows.
> 
> Mmm, I see. It seems strange to me, because when it gives me the recovery option and I enter the BIOS, it boots normally and then It lets me start Windows from Limine. So it's a Windows problem, It detects that Windows didn't boot correctly and therefore launches into recovery mode. But I think what I'll do is create a script that starts Windows bypassing the chain part.

I'm having the same condition today. My dual boot had been working perfectly for 6 months until today. I'm still trying to find a fix.