Limine-Bootloader/Limine #427: fallback initramfs entry?

wastlnd, 2024-11-02:
I am new to limine but with the help of the Arch Wiki I have successfully setup a main Linux entry. My question is: is it possible to add a fallback initramfs entry like in the GRUB "Advanced options"?And if yes, how? TIA.

faiz-x, 2024-11-04:
Set module_path to fallback-initrd  (which is typically present in ESP)

wastlnd, 2024-11-04:
@faiz-x ok, so I add something like`/Arch 6.6 fallback` and leave the cmdline unchanged, just point the module_path to the fallback initramfs? Also, what's the  difference between limine and linux as protocol?

faiz-x, 2024-11-04:
> I add something like`/Arch 6.6 fallback` and leave the cmdline unchanged, just point the module_path to the fallback initramfs?

Yes.

> what's the difference between limine and linux as protocol?

You can read about the limine protocol [here](https://github.com/limine-bootloader/limine/blob/v8.x/PROTOCOL.md)

wastlnd, 2024-11-04:
Thank you, I tried to use limine but only linux worked.

mintsuki, 2024-11-07:
That is normal. The `limine` boot protocol needs to be supported by the kernel, and Linux only supports the `linux` protocol.

closed 2024-11-07 by mintsuki