Limine-Bootloader/Limine #441: Triple fault during kernel loading on ASUS UX303L

apache-hb, 2024-12-27:
## Laptop 

| Component | Name |
| ------------- | ------ |
|Model | ASUS UX303L
|BIOS | American Megatrends v204 (2.15.1236) Released 2012
|GOP Version | 5.0.1036
|EC Version | B01S090301
|Processor | Intel(R) Core(TM) i7-4510U CPU @ 2.00GHz
|Total Memory | 8192 MB
|Intel Virtualization Technology | [Enabled]
|Intel AES-NI | [Enabled]
|Fast Boot | [Disabled]
|Launch CSM | [Disabled]
|Secure Boot Control | [Disabled]
| Boot Mode | UEFI Only |

## USB Drive
| Component | Name |
| ------------- | ------ |
| Name | Netac OnlyDisk USB Device 64GB

I copied `template-x86_64.iso` onto the drive using rufus in ISO mode, but observed identical behaviour with DD mode as well as flashing from a linux host onto the USB drive.

## Limine
| Component | Name |
| ------------- | ------ |
| Limine | prebuilt from `git clone https://github.com/limine-bootloader/limine.git --branch=v8.x-binary`
| Template | `https://github.com/limine-bootloader/limine-c-template.git` cloned 26-12-2024

## Expected behaviour

A kernel is loaded and executes.

## Actual behaviour

Limine displays the boot prompt and all other menus successfully. But when loading a kernel the system resets.

Tested with limine-c-template cloned 26-12-2024 and my own kernel, neither of which loaded.

<details>
<summary>Build log</summary><p>

```bash
elliothb@ELLIOT-SERVER:~/github$ git clone https://github.com/limine-bootloader/limine-c-template.git
Cloning into 'limine-c-template'...
remote: Enumerating objects: 688, done.
remote: Counting objects: 100% (214/214), done.
remote: Compressing objects: 100% (128/128), done.
remote: Total 688 (delta 154), reused 134 (delta 83), pack-reused 474 (from 1)
Receiving objects: 100% (688/688), 91.48 KiB | 1.41 MiB/s, done.
Resolving deltas: 100% (420/420), done.
elliothb@ELLIOT-SERVER:~/github$ cd limine-c-template/
elliothb@ELLIOT-SERVER:~/github/limine-c-template$ gmake all
rm -rf limine
git clone https://github.com/limine-bootloader/limine.git --branch=v8.x-binary --depth=1
Cloning into 'limine'...
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 19 (delta 6), reused 11 (delta 6), pack-reused 0 (from 0)
Receiving objects: 100% (19/19), 797.20 KiB | 2.42 MiB/s, done.
Resolving deltas: 100% (6/6), done.
gmake -C limine \
        CC="cc" \
        CFLAGS="-g -O2 -pipe" \
        CPPFLAGS="" \
        LDFLAGS="" \
        LIBS=""
gmake[1]: Entering directory '/home/elliothb/github/limine-c-template/limine'
[... 132 lines cut ...]
Added to ISO image: directory '/'='/home/elliothb/github/limine-c-template/iso_root'
xorriso : UPDATE :      11 files added in 1 seconds
xorriso : UPDATE :      11 files added in 1 seconds
ISO image produced: 2014 sectors
Written to medium : 2014 sectors at LBA 0
Writing to 'stdio:template-x86_64.iso' completed successfully.

./limine/limine bios-install template-x86_64.iso
Physical block size of 512 bytes.
Installing to GPT. Logical block size of 512 bytes.
Secondary header at LBA 0x1f77.
Secondary header valid.
GPT partition NOT specified. Attempting GPT embedding.
New maximum count of partition entries: 44.
Stage 2 to be located at 0x3e00 and 0x3eac00.
Reminder: Remember to copy the limine-bios.sys file in either
          the root, /boot, /limine, or /boot/limine directories of
          one of the partitions on the device, or boot will fail!
Limine BIOS stages installed successfully!
rm -rf iso_root
elliothb@ELLIOT-SERVER:~/github/limine-c-template$ 
```

</p>
</details>

apache-hb, 2024-12-27:
Video demonstrating the issue.

https://github.com/user-attachments/assets/52cbbfb3-9c32-4d64-bb53-d4208af29e44

apache-hb, 2024-12-27:
I've narrowed down the reset to somewhere in these lines 
https://github.com/limine-bootloader/limine/blob/v8.x/common/protos/limine_32.asm_x86#L76-L92

apache-hb, 2024-12-29:
More testing narrowed it down to the `call .p2` instruction

apache-hb, 2025-01-04:
Some more discussion happened on discord, these are the important bits:
* Limine produces a very strange memory map, inspecting the UEFI memory map with the UEFI shell is mostly normal.
* The memory map changes between system resets.

Possible issues are either the stack pointer being incorrect, or the memory mappings for the higher half being incorrect.

iamdavidcz, 2025-02-17:
I am observing that same behaviour on my ARM64 UEFI laptop (Acer Aspire 1 A114-61) - Linux kernel does not boot and there is infinite bootloop. If I comment out `efi_exit_boot_services();` line I am getting: `failed to set UEFI virtual address map 0x3`, so I suspect incorrect memory map too.

GRUB boots the same kernel without any issues.