Limine-Bootloader/Limine #445: On x230 limine doesn't load if CSM is turned on

programmerlexi, 2025-01-04:
On a Thinkpad x230 limine does not load if CSM is turned on.

Basically the inverse of #417.

mintsuki, 2025-01-04:
Hello, thanks for reporting this. Do you have any other info to add? Like precised firmware settings used, the way Limine was installed on the boot medium, etc.

programmerlexi, 2025-01-04:
Limine was installed on a USB.
Limine 8.6.1 from nixpkgs was used.

Firmware Settings:
- UEFI only
- CSM on
- Boot mode is set to quick (Diagnostic doesn't visually change anything)

On boot only a black screen is shown.

programmerlexi, 2025-01-04:
Legacy only with CSM on works fine.

programmerlexi, 2025-01-04:
Secure boot is disabled

programmerlexi, 2025-01-04:
The bios is on an old version (2.50)

mintsuki, 2025-01-04:
How was it installed on the USB?

programmerlexi, 2025-01-04:
By making an iso using xorriso and putting that on the USB.

```bash
xorriso -as mkisofs -b boot/limine/limine-bios-cd.bin \
    		-no-emul-boot -boot-load-size 4 -boot-info-table \
    		--efi-boot boot/limine/limine-uefi-cd.bin \
    		-efi-boot-part --efi-boot-image --protective-msdos-label \
        	iso_root -o image.iso
```

mintsuki, 2025-01-04:
Did you run `limine bios-install image.iso` afterwards on the created image?

programmerlexi, 2025-01-04:
yes

mintsuki, 2025-03-05:
This may be addressed by the same work-around as #470 

Please feel free to reopen if necessary.

mintsuki, 2025-03-05:
I reopened that issue as the workaround was premature and not reliable. Also it was unlikely to be the cause of this, on second read.

mintsuki, 2025-03-06:
Scratch the above comment. I reapplied the fix and it should indeed work around your issue as well. Please reopen if this isn't the case.

closed 2025-03-06 by mintsuki