Limine-Bootloader/Limine #455: Limine Providing 0 framebuffers

ApplePieCodes, 2025-02-03:
Everytime i boot my kernel (with qemu-system-x86_64 lithium.iso -bios OVMF.fd), limine provides 0 framebuffers.
My kernel repo is https://github.com/[ApplePieCodes/lithium](https://github.com/ApplePieCodes/lithium).
I have also tried the official limine c template, replacing my kmain with the limine c template one. Is this a limine issue?

el-zorab, 2025-02-04:
@ApplePieCodes I cannot reproduce at your last commit. Flanterm successfully prints `Hello, World!` to the screen, therefore a framebuffer is passed to your kernel correctly. What makes you think Limine passed 0 framebuffers?

marv7000, 2025-02-04:
Have you tried on real hardware? If no, what is your QEMU and OVMF?

ApplePieCodes, 2025-02-05:
> [@ApplePieCodes](https://github.com/ApplePieCodes) I cannot reproduce at your last commit. Flanterm successfully prints `Hello, World!` to the screen, therefore a framebuffer is passed to your kernel correctly. What makes you think Limine passed 0 framebuffers?

As of Yesterday, I fixed the problem. It was a problem with the segments in my linker.ld.

closed 2025-02-05 by ApplePieCodes