Limine-Bootloader/Limine #456: Config validation

tribals, 2025-02-13:
It would be nice to have config validation in `/usr/bin/limine` so one can know that it is not valid _before_ he will run into trouble, especially if there is already `enroll-config` sub-command one need to run if he uses Limine with secure boot enabled.

😊

mintsuki, 2025-03-03:
Sorry, this feature is not planned.

closed 2025-03-03 by mintsuki

tribals, 2025-03-04:
No problem - I stopped using this bootloader.

boomshroom, 2025-03-05:
"Not planning" a method to convince users that they didn't mess up seems like an odd decision. Obviously it can't catch everything, but it can at least confirm that the file *parses* without needing to install a VM just to tell.

mintsuki, 2025-03-05:
There are a million ways one can screw up a config file such that it renders the system unbootable, especially when runtime config editing is disabled, and it is simply not possible to catch them all ahead of time. Think things like wrong kernel parameters, wrong file paths... (both of which are *extremely* common, way more than screwing up the syntax of nested config entries).

Adding this seems like it would only add some false sense of security if it were to say a config is "valid" just for it to not boot due to one of the many other issues. It's a half measure.

The file you linked is an example of where Limine may perhaps be a bit more lenient and try to work around the issue instead of relying on a soft panic (which in any case still gives you the option to edit a blank entry manually if the editor is enabled). But that is a different issue that can be solved separately.

I hope this explains the logic behind not planning this feature.

tribals, 2025-03-05:
"Wrong command line" etc going beyond incorrect config file syntax. The point was that _if_ enroll config is already employed in reading that file - then why not it can make sure it at least parses correctly? This is especially true in spite of changing such syntax. Is that change will be done again? Why it was done?

But nevermind - I don't care. Just curious why such a simple feature isn't already implemented. May it be because of we should stop installing useless bootloaders re-inventing the wheel, as most systems nowadays already contains a bootloader - that is, EFI bootloader.

mintsuki, 2025-03-05:
Because it is not "such a simple feature", it still involves adding config parsing code to the `limine` host executable that does not have it (and no, `enroll-config` does not require parsing and I would advise you to read the code before making stupid comments like that), and the parsing code would need to be different than the Limine bootloader code itself.

Is it doable? Yeah sure, at least to validate basic config syntax. Is it worth it? Strong no, in my opinion, due to the aforementioned reasons.

In any case I shouldn't even be answering you due to the lack of respect and rudeness displayed here, but I'd rather still express myself for anyone finding this issue in the future. You're welcome to not use "useless bootloaders".

boomshroom, 2025-03-06:
> I hope this explains the logic behind not planning this feature.

I may personally disagree with the conclusion, but yes it does.