Limine-Bootloader/Limine #460: AArch64 Linux fails to boot on Acer Aspire 1 A114-61

qookei, 2025-02-18:
> I am observing the same behaviour on my ARM64 UEFI laptop (Acer Aspire 1 A114-61) - Linux kernel does not boot and there is infinite bootloop. If I comment out `efi_exit_boot_services();` line I am getting: `failed to set UEFI virtual address map 0x3`, so I suspect incorrect memory map too.
> 
> GRUB boots the same kernel without any issues.
> 
> Edit: I can reproduce this issue in 9.0.0. 

 _Originally posted by @iamdavidcz in [#441](https://github.com/limine-bootloader/limine/issues/441#issuecomment-2664027574)_

iamdavidcz, 2025-02-19:
@mintsuki @qookei

If you need any logs/more tests I am more than happy to help.

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.

closed 2025-08-31 by qookei