Limine-Bootloader/Limine #470: BIOS Limine does not load on T460p

programmerlexi, 2025-03-04:
Limine version tested: 8.7.0, 9.0.1
Source of the package: nixpkgs

Limine does not load on my ThinkPad T460p, when set Legacy Boot Only.

It just flashes a black screen and returns to the boot menu.

lukflug, 2025-03-04:
I was able to reproduce the issue on the ThinkPad T480.

lukflug, 2025-03-05:
It seems to be related to Lenovo's CSM implementation refusing to boot GPT partitioned media via CSM. The boot sector doesn't even seem to be executed before it returns to the boot menu. Booting off of MBR partitioned media with "Legacy Boot Only" set does work.

JohnRTitor, 2025-03-05:
> It seems to be related to Lenovo's CSM implementation refusing to boot GPT partitioned media via CSM

So it ultimately depends on the CSM implementation in the firmware. @programmerlexi have you checked with systemd-boot?

mintsuki, 2025-03-05:
Hello. So, as addressed in the commit referenced, this is caused by BIOS/CSM systems that do not like to boot from media with a GPT... it sucks but it is what it is.

If one follows that note to convert the GPT to an MBR, the issue should be fixed (or worked around, really).

Please let me know!

mintsuki, 2025-03-05:
The issue should be mitigated by 70d7b202. This forces the protective MBR partition to have the bootable flag set, even on GPT. The issue this wasn't done before, I believe, is reports of it causing some *other* machines to have issues booting...

I will investigate and let you know.

mintsuki, 2025-03-06:
I reapplied the original changes from cd51de2 and reverted 70d7b20. This seems to be the setup that has the most wide compatibility of all, and the issues I originally reverted cd51de2 for cannot be reproduced widely.

commit cd51de29 (2025-03-05): Revert "docs: Address issue with ISOHYBRIDs and some BIOS systems. Closes #470"

This reverts commit 2ac13902dd1abcc66d1da0c776adfea8249df5cd.
commit 9f937d95 (2025-03-06): docs: Address issue with ISOHYBRIDs and some BIOS systems. Closes #470
commit 65d43f56 (2025-03-07): Revert "docs: Address issue with ISOHYBRIDs and some BIOS systems. Closes #470"

This reverts commit 9f937d95559c2ab0c19a0e50588be52938c92ec2.
closed 2025-03-06 by mintsuki