Limine-Bootloader/Limine #473: Support systemd Boot Loader Interface

lzcunt, 2025-03-13:
[The systemd Boot Loader Interface](https://systemd.io/BOOT_LOADER_INTERFACE) is an interface between the operating system and the boot loader, making use of EFI variables. It can identify the boot loader to the operating system, provide boot performance metrics, information about entries, and a set-on-install random salt to be combined with the RNG seed if the bootloader provides a RNG seed to the kernel (useful for cases where disk images are copied across systems, resulting in the same RNG seed file to be used across different systems - the machine-specific salt stored in an EFI variable will be combined with the RNG seed to generate a unique RNG seed for each system for in case this happens). EFI variables can be set by the operating system to instruct the boot loader to pick a different entry or skip the menu. It also can tell the OS whether the Boot Loader Specification #472 is supported by the boot loader, whether the boot loader provides a RNG seed to the kernel, or whether Automatic Boot Assessment is supported.

I'm not knowledgeable enough about this so the above description may be inadequate or partially wrong.

LordKeravnos, 2025-03-13:
Please forgive my ignorance as I am neither a Limine maintainer nor well versed in the inner workings of it, but why is this necessary? Limine already works fine with every distro regardless of systemd or not (I personally run Limine on OpenRC Gentoo, and on a laptop with dinit Artix) and the configuration is identical regardless of init/service manager. Supporting this sounds to me pointless and bloated, especially in a project like Limine which focuses on being lightweight, as well as supporting more than just Linux.

I am seriously curious as to why you suggest this should be supported especially when you yourself say you're not knowledgeable enough - doesn't that alone make this proposal a waste of maintainers time?

lzcunt, 2025-03-13:
> Please forgive my ignorance as I am neither a Limine maintainer nor well versed in the inner workings of it, but why is this necessary? Limine already works fine with every distro regardless of [...]

While this specification comes from systemd, it's neither systemd nor Linux specific. If you're using systemd-boot or GRUB with the necessary patch, you can query e.g. boot times through EFI variables on your OpenRC and dinit machines too. And the specification allows picking which variables are supported and which are not, so each feature can be discussed on a case-by-case basis. In fact, most of it is so simple, the patchset to implement this for GRUB seems to be ~200 lines (the more complicated ones are complicated only because they depend on other features to be implemented by the bootloader, like reading XBOOTLDR according to the Boot Loader Specification not to be confused with this Boot Loader Interface spec). TL;DR: This is a feature systemd makes use of and graciously standardizes for everyone (including non-systemd) to make use of.

> I am seriously curious as to why you suggest this should be supported especially when you yourself say you're not knowledgeable enough - doesn't that alone make this proposal a waste of maintainers time?

I am not knowledgeable enough simply because I was made aware of it just today, by someone from the lanzaboote community. In fact, this seems simple enough compared to the other two proposals I opened today that I am down to implement and maintain this feature (possibly in the summer, very busy until then)

LordKeravnos, 2025-03-13:
This is not a lot I can comment on, as I was only curious and of course don't know enough about Limine's inner workings myself, but considering XBOOTLDR features have been rejected as per #346 I don't see a lot of this becoming reality, if any of it. This seems like a very Linux-centric thing for a bootloader that doesn't focus exclusively on Linux, but again you'll have to wait for a reply from a maintainer for that. Thank you for replying and sating my curiosity.

lzcunt, 2025-03-13:
> This is not a lot I can comment on, as I was only curious and of course don't know enough about Limine's inner workings myself, but considering XBOOTLDR features have been rejected as per #346 I don't see a lot of this becoming reality, if any of it.

I've opened these 3 issues seperately because while Automated Boot Assessment seems to depend on the other two, Boot Loader Specification and Boot Loader Interface suggest each other but do not *require* each other. This is entirely possible to support without Boot Loader Specification support.

> This seems like a very Linux-centric thing for a bootloader that doesn't focus exclusively on Linux,

Considering Managarm is steadily moving towards systemd on Managarm, this is not necessarily a Linux-centric thing (so far it is). I have not discussed this with Managarm developers and their time is probably spent better shaping and stabilising their systemd port so don't bother them about it yet IMO.

Even if it is Linux-centric, lanzaboote (awesome tool from the NixOS community for secure boot on NixOS) does depend on it. Limine support in lanzaboote is in the discussion stage and this is needed by lanzaboote's UKI stub. See [this](https://github.com/nix-community/lanzaboote/issues/444#issuecomment-2721718432). It would probably benefit other distributions and tools as well.

> ... but again you'll have to wait for a reply from a maintainer for that.

~~Yes, mint's word is law here~~

> Thank you for replying and sating my curiosity.

Np. I hope to see Limine become a first class citizen for secure and modern boot in NixOS and other distributions.

closed 2025-03-15 by mintsuki