Limine-Bootloader/Limine #480: limine-snapper-sync: Fatal error compiling: error: release version 17 not supported

BoardWatcher, 2025-03-24:
I am trying to install liminie-snapper-sync with pacman.  During the clean install process I obtain the error below.  Please advise.
Note: I'm using Arch Linux 6-13 with Java 11 JDK installed.

`[✖] 󰛓  java -version
openjdk version "11.0.26" 2025-01-21
OpenJDK Runtime Environment (build 11.0.26+4)
OpenJDK 64-Bit Server VM (build 11.0.26+4, mixed `mode)``

`[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.13.0:compile (default-compile) on project limine-snapper-sync: Fatal error compiling: error: release version 17 not supported -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.13.0:compile (default-compile) on project limine-snapper-sync: Fatal error `compiling``

lzcunt, 2025-03-24:
limine-snapper-sync is third party and not developed here. Please reach its maintainer(s) instead. That looks like you're building with the wrong Java version to me, but again this is not an issue with Limine, please reach limine-snapper-sync's maintainer.

mintsuki, 2025-03-24:
Yeah, as above. `limine-snapper-sync` is a 3rd party utility not developed by us.

mintsuki, 2025-03-24:
But just in case, I'll ping @Zesko as I believe they are the author.

closed 2025-03-24 by mintsuki