Limine-Bootloader/Limine #482: multiboot2: Canoot boot Xen Hypervisor multiboot kernel

digitalrane, 2025-03-29:
When testing booting the Xen Hypervisor under Limine, in both BIOS and UEFI modes, using the `multiboot2` module, the following fault is looged, and boot does not proceed:

`multiboot2: required EFI AMD64 entry tag is unsupported and no alternative entry was found`

This currently is on [this line of code in the multiboot2 module source](https://github.com/limine-bootloader/limine/blob/trunk/common/protos/multiboot2.c#L218). This code appears to be essentially a final attempt to find an entry point.

I've looked over the Xen sources to see when tags/symbols are published in the multiboot2 header, and it does seem that the EFI AMD64 entry tag is present, but maybe with a name or index not expected by Limine. However, it's also possible the required information should be picked up by another tag of field in the header which is not being detected. GRUB2 for example is able to boot this kernel using its multiboot2 support, as multiboot2 is the suggested way Xen be booted and many of the documented examples for both BIOS and EFI booting use Grub2 configuration files.

The main structure of the Xen kernel's multiboot2 header for x86_64 is [defined in this file in the Xen source](https://xenbits.xen.org/gitweb/?p=xen.git;a=blob;f=xen/arch/x86/boot/head.S;hb=HEAD).

This testing was performed on NixOS as part of developing and testing BIOS and EFI booting capabiliies in [this PR](https://github.com/NixOS/nixpkgs/pull/393287)

Some related references on how Xen boots, and multiboot configuration (mentions the `start` symbol being used):
 - https://xenbits.xen.org/docs/unstable/hypervisor-guide/x86/how-xen-boots.html

mintsuki, 2025-03-29:
Limine does not support booting multiboot2 kernels that expect to be loaded solely via the EFI handoff method. Now, this would be a bug if the kernel does in fact provide an alternative non-EFI handoff entry point, and Limine somehow does not detect it...

If the kernel only has an EFI handoff entry point then I don't see how using multiboot2 to load it would at all work on BIOS-only systems... So it either that configuration is not supported, or Limine has a bug and we're missing the non-EFI entry point.

Is there a chance you could provide the file you're trying to boot?

digitalrane, 2025-03-29:
@mintsuki sure thing, I've uploaded a kernel bzImage and Xen multiboot image build on NixOS which I was using to test this out. The Xen multiboot kernel is likely the useful file here, as that should give you an insight into which symbols are available/not available. https://junkyard.systems/limine-nixos-test-binaries.tar.xz

mintsuki, 2025-03-31:
Okay, so this should be fixed by c8cdc9e05125fadca8ab0740387e8d4b33ad52c8 and 80912a3d1fb16c384b9781a91c35a99ad8929768. Let me know if there are further issues related to this.

closed 2025-03-31 by mintsuki