Limine-Bootloader/Limine #487: Automatic boot does not work for subentries

ventureoo, 2025-03-30:
Hi, a small non-critical issue I'd like to report. If the entry is at the top level, autoboot and saving works for it, but for subentries autoboot even with default expanding via ``/+`` does not work.

Probably related with https://github.com/limine-bootloader/limine/issues/459.

mintsuki, 2025-03-31:
Hello, I can't quite reproduce this... Could you share the config you used to reproduce this?

mintsuki, 2025-03-31:
If you don't have a `default_entry` option set, the default entry will be 1, which in this case is a directory rather than an entry proper. When the default entry is set to a directory, Limine will disable autoboot. This is basically intended behaviour.

closed 2025-03-31 by mintsuki

ventureoo, 2025-03-31:
> If you don't have a `default_entry` option set, the default entry will be 1, which in this case is a directory rather than an entry proper. When the default entry is set to a directory, Limine will disable autoboot. This is basically intended behaviour.

I've tried this before, but it only seems to work in combination with the default unexpand via ``/+``. Thanks and sorry for the inconvenience!